fesch/CanZE #16: Request to fesch: throttled listeners

yoh-there, 2015-09-17:
May I suggest to add an optional parameter "cycles" to addListener, being an integer, stating per how many cycles this field should be actually queried. Rationale is that i.e. cell temperatures vary very slowly and need to be queried only once per, say 10 cycles. This will speed up the updateing of more fluid fields.

fesch, 2015-09-17:
Okay, I see what you mean. The question that I'm asking myself know is, if this is not rather something related to the "field" itself?

Actually it's the "device" that loops querying the fields from the ELM simply one after the other. The "device" does not know about the listeners connected to the fields, but if the information would be stored in each field (e.g. come statically from our spreadsheet), I think the Device-Class could implement some king of scheduler instead of looping the fields around one after the other.

yoh-there, 2015-09-17:
Yes, it could be defined in the spreadsheet (ending up in Field(s)), however, I can imagine one screen where Field x would require a fast update, and and another where the Field can be updated less frequently. So I am not so sure.

It is somewhat related to the time outs I would think. Those should definitely be in the Field(s).

yoh-there, 2015-09-19:
I have implemented this in Fields, Field and ELM, but it doesn't work yet in ELM. Haven't figured out why yet.

fesch, 2015-09-20:
Yes, the problem was, that the code that switches to the next filter was enclosed in the if that tells the filter to run, so the fieldIndex was never incremented. This should work now.

yoh-there, 2015-09-20:
Thank you! I was searching in cloning. 4 eyes...... :-) Will pull and test soon.

fesch, 2015-09-23:
Implemented by yoh-there through the "skips"

closed 2015-09-23 by fesch