fesch/CanZE #19: Bug in payload bitcount handling for SINGLE ISO-TP frames

yoh-there, 2015-09-18:
Bug found (and fixed) in the ELM driver, that removed the first payload byte. Probably caused my miscounting startbit for single frame UDS messages, while non-UDS messages are very often FIRST/NEXT (where the bug AND the miscounting was not present.

While the issue is closed in the code, it needs to be fixed in the spreadsheet and double checked in the initialisation of the Fields class, so I am leaving this open.

yoh-there, 2015-09-20:
AFAIK fixed throughout the code.

fesch, 2015-09-20:
Did you make a push?

yoh-there, 2015-09-20:
To yoh-there/CanZE (yesterday): yes. Pulling into fesch: no. I haven't dared doing that to be honest.

yoh-there, 2015-09-20:
This has been fixed now. Issue closed.

closed 2015-09-20 by yoh-there