fesch/CanZE #21: Request to add brake pedal position to driving screen

yoh-there, 2015-09-20:
Request to add brake pedal position to driving screen.

NacGit, 2015-09-21:
When you use the break pedal it would very useful to know if you are using only the regeneration motor as a break or also the standard mechanic break.

yoh-there, 2015-09-21:
I understand, but here is my, ahem", analysis. You couldn't know as it is in the spreadsheet. The Renault mechanic I spoke to seems to disagree with me.

I have put some time in understanding the braking system. While I am not absolutely sure, it seems to me (and contrary to Renault marketing fluff) the brakes are fully traditional power-assisted brakes (with a servo instead of a vacuum system of course), but friction braking is NOT being disengaged assuming the same pedal position when regenerative braking is available. Of course, regenerative braking does increase when pushing the pedal, but that action is implemented mainly in the "first part" of the pedal push. From an energy conserving point of view, the preferred optons seem to be:
1. coasting in N (just natural losses)
2. coasting in D for more braking (add regenerative load)
3. push pedal in N (add friction load and engage friction)
If correct, this implies there is no way to let teh car regenerate 43kW while NOT engaging the friction brakes. This in turn implies there is no "smart" way of braking, other than not pushing the pedal at all, suggesting an app screen simply doesn't make sense.

NacGit, 2015-09-21:
Ok, I understood from you analysis that the friction brake is always engage, something logical from the security point of view.
But here is my experience with the car.
There is a point in the break pedal that only the regenerative break and not the friction brake is working.
Could be possible that at this point of pressure over the pedal, the break pads aren't in contact with the break disk, or not with the needed force to be aware of it.
But in a practical point of view, if there is no signal in the obd2 connection about that, nothing we can do.
By the way. thanks for all your efforts !!!

yoh-there, 2015-09-21:
Thank you for your compliments.
What you are asking is a "red light" whenever the friction brakes are being engaged. I get that. I have found "Main Cylinder Pressure" in the i907, which would expect to correspond with that, but what I have seen, the brake fluid pressure goes up almost immediately when pushing the brake pedal. You can try the following experiment: drive on a flat road, set the gear in N and push only a little bit. You will feel the friction brakes engaging, though lightly, almost immediately. Now keep your foot stead and gear to D. You will feel the regen kicking in. But when doing that, I have not seen ANY decrease in the main cylinder pressure. Only if "Main Cylinder pressure" is just a servo signal to the ABS/regen system, there could be "smarter breaking" than what I have described before, but you know what? It goes op to almost 100 bar :-o For a servo signal? I don't think so, unless the i907 has labeled that field in error.

yoh-there, 2015-09-22:
New information! It seems we have a field called "FrictionTorque". If that means what I think it means, we have that "red light", and I can also test my thesis above. So much to do, sol little time. Stay tuned!

fesch, 2015-09-22:
The field (as defined in the sheet) has an initial value of 600. If the car decelerates, breaking or not, the values increases. If the only the mechanical break is in action, its value is again 600 ...

Does that make your "red button"?

yoh-there, 2015-09-22:
So you think that value means stricktly the Motor Breaking (regen) part? If so, weird naming!

fesch, 2015-09-22:
Yes. Using the mechanical break (< 10km/h) does not make this value change.
But maybe there is some other field that we can use for your red button. I've already found, that some fields are not present of without meaning on the Zoé. I suppose the document to have a common base for all their car. A field like "WaterInDieselError" does not make any sens on a VE too ;-)

yoh-there, 2015-09-22:
Hmmmm. In that case, need to search deeper.

yoh-there, 2015-09-27:
Current fields displayed IMHO do not give a solid indication when the friction brakes are applied. Hmpf

yoh-there, 2015-09-28:
I am trying to understand the raking system. Observations so far.

0x130-20  is literally an active REQUEST. Regen while not touching the brake is NOT in this number. I need to figure out if it is completely passive (value 4094) when coasting OR braking when in gear Neutral, which would be logical. What is interesting is that the requested Torque goes to neutral when the speed goes to 0

0x130-44 seems to correspond with the brake pedal position percentage (roughly 20 Nm per % pedal)

0x18a-27 seems to be a total breaking torque, but I am not sure yet. It definitely includes the regen when coating without breaking. I think it does not include friction breaking.

yoh-there, 2015-09-29:
More info:
130-20 is indeed the requested regen torque EXCLUDING coasting
18a-27 is in my opinion ONLY the regen torque for coasting

In theory, and when the calibration is correct, this means that True friction could be (torque requested by driver) minus (torque requested to motor) minus (coasting torque).

yoh-there, 2015-10-02:
For the time being I am closing this issue. Braking is complicated and will remain in a separate screen. Once we are sure we can deduct the torque of the friction brakes, we might decide to add a simple "waste" indicator on the driving screen.

closed 2015-10-02 by yoh-there

yoh-there, 2015-10-04:
I have added brake pressure (free frame 352,24) in a progress bar, and I think I know now what is happening. Please verify.

- Drive brake wheels torque request (2nd bar) is the drivers braking action (pedal), translated to torque. I am reasonably sure this is total braking torque requested.
- Electrical Brake Wheels Torque is the braking (regen) torque the motor is supposed to deliver. It goes down to 0 when speed closes to 0, which makes sense as the motor cannot regen close to standstill.
- Coasting Torque (Called FrictionTorque in some sources) is the braking torque applied by the motor when driving and releasing the accelerator pedal. 

Therefore I believe the total torque coming from the motor (regen) is the sum of the first and third bar, and maybe I should simply make it one bar in the display.

The Brake Pressure (not a torque) seems to correspond with the actual friction brakes. It also seems, like I suspected, that they are not engaged in the first aprx 25% of the brake pedal movement, and after that it will simply assist the motor brake.