fesch/CanZE #23: Battery compoartment temperatures not right on Fluence ZE

yoh-there, 2015-09-20:
[moved from another issue thread]
=======
solmoller:
(...) And I rarely get a sane temperature reading on the charging screen

======
yoh-there:
The sane part come form the Fluence ZE having only 4 compartment sensors instead of 12 and I have not made Zoe/Fluence variants. Your first 4 values should be "sane", but I would love to hear if the ARE sane!

=======
solmoller:
I've had sane results from sensors 7 and 12, but I think that it could be just random numbers now.. Any suggestions on how to dig out Fluence battery temps? Henrik

=======
yoh-there:
Sure! According to Alexandre, the Fluence's temperatures are given with request https://docs.google.com/spreadsheets/d/1p_zEgy8RqvTV79mT4HFdyKinz3l3q-WlECp_W2ymePM/edit#gid=651307834 79b/7bb, request 0x2104/0x6104 (same as Zoe), bit positions 32, 56, 80, 104, and an average in 12 Not that this is the same as the first 4 temperatures in the Zoe, so i would expect the first 4 to be right, and the next 8 to be nonsensical. Somebody should really integrate Alexandre's work in the main spreadsheet. What would be great if we then create two initialization variants of the Fields class. Note that there is a "car" column (Fluence/Zoe/Both) in the spreadsheet, so it is tedious, but not hard.

=======
fesch:
That right. I've already prepared the settings activity to let the user chose his "car" (Zoe or Fluence for now), so this deployment should not be that hard.

=======
Meanwhile i found the same - but I am not sure we draw the same conclusion: Zoe Battery T1 I read as 7bb.6104.32, Fluence I'd presume is 7bb.2104.32 Will test shortly

=======
yoh-there:
Nope, 21 is the request code, 61 the answer. The app uses only the reply code and uses internal tables to send the proper request.

=======
solmoller:
Ah, never mind, I see in the Fields.java, that it may be me misreading the spreadsheets

(...)

Oh, Henrik, I think I found it. For the Zoe, the temperatures have an offset of 40, so the real values can go from -40 to 215 ;-) The Fluence ZE does not have those offsets. I was looking at your values (your email of sept 14) and it completely makes sense. At that moment, your 4 compartment temps were 21, 20, 19, 19. So yes, we do need to work in the initialisation of Fields.

yoh-there, 2015-09-20:
I will add the proper fields to the spreadsheet coded for Fluence ZE only. In essence this issue is a copy of #20.

ps: Fesch, be careful please, as I inserted a new column in the static data.

fesch, 2015-09-20:
What did you add?

yoh-there, 2015-09-20:
FIELD_SKIPS, and corresponding stuff in Field Class. Still doesn't work in ELM, but that's another matter.

fesch, 2015-09-20:
Push, push ... :-) Actually I can't see it :-/

yoh-there, 2015-09-20:
It is there!!! In yoh-there, I pushed it yesterday and issues a pull
request????

yoh-there, 2015-09-20:
https://github.com/yoh-there/CanZE/blob/master/app/src/main/java/lu/fisch/canze/actors/Fields.java

ie line 29

fesch, 2015-09-20:
I did not get any pull request :-(
Anyhow, I see it there. I will add a column to the spreadsheet to, as I'm exporting data from there.

fesch, 2015-09-20:
The app now let's the user chose a "car" and will request other data based on this information.

We need a Fluence user to test this too. For example the Cell-Temperature-Screen. It works fine on my Zoé ...

yoh-there, 2015-09-20:
fesch imported the car column in "Fields" now. Wouldn't it be a better approach to select on initialization of the object (fillStatic), to only add the generic fields and the one correspondign to the current car? I think in the current implementation it is quite hard to select the proper filter when the requests between cars are identical but actual handling is not (see compartment temperature)?

Or do you propose to solve that problem in a different way?

fesch, 2015-09-21:
Applying the car filter during the initialization of the Fields class is a bad idea, as the app will crash in some cases. Let's take as example the CellTemperatureActivity. If you chose "Zoé", all 12 fields will be loaded and display correctly, but if you chose "Fluence", only 4 of the fields will be loaded and then the app will crash when opening the activity, as the 8 remaining fields do not exists.

My approach is to use the car combined with the SID to find the correct field. In the way I did it, it is transparent to the API user. On an activity that has been designed for a specific car, users that do not have configured THIS car, will see incorrect values (of course), but the app will remain consistent.

solmoller, 2015-09-21:
..or we remove fields that are not populated in the GUI, that was my plan

Henrik

fesch, 2015-09-21:
Yes, that is also possible ... and even both methods can be implement simultaneously as each field has an indicator for what car it is.

yoh-there, 2015-09-21:
I modified driver and charging to at least not crash on an invalid addListener parameter. Since no listener is added, no queries are performed, and no invalid data comes back.

yoh-there, 2015-09-21:
Car selector does not work very well yet. The battery compartment temperature now uses Fluence for the first 4, and Zoe for the next 8. Happened for both Driving, Charging as well as BatteryTemperature.

fesch, 2015-09-21:
This morning it worked on my Zoé ... 
Did you go to settings and change the car there? (I suppose you did, but just to make sure ...)

yoh-there, 2015-09-21:
It worked, but first 4 compartments displayed +40 C ;-) Car is set to Zoe, but maybe that is an unsaved default? When I drive home end of the day, I will make sure I change to Fluence and back to Zoe, save in between, to make absolutely 100% sure. Will post result here.

yoh-there, 2015-09-21:
Sorry no, same result. I will dig through the code after dinner.

yoh-there, 2015-09-21:
I see the problem I think. Fluence = car 1 in the spreadsheet, 2 in the code and vice versa. I did that out of respect, Fluence ZE was born earlier ;-) Fixed it on the constants, comitted, pushed and issued a pull request.

@Henrik, at the moment in driving and charging, I always set default values in the screens, it helps me debugging. It the 8 useless #T fields annoy you, we can simply change the default in strings.xml.

yoh-there, 2015-09-21:
ps: I think 

        // if none is found, try the other one, starting with 0 = CAR_ANY
        for(int i=0; i<3; i++) {
            tryField = fieldsBySid.get(i + "." + sid);
            if (tryField != null) return tryField;
        }

is not the proper strategy. If the car specific filter is not found, IMHO only the generic ("0") should be tried, not just any car.

henrik, 2015-09-21:
Wrong Henrik, unsubscribing :)

yoh-there, 2015-09-21:
oops, sorry. I meant @solmoller

fesch, 2015-09-21:
No, I think that does not work, because if no field is found, the app would terminate with a NullPointerException, which I want to avoid.

I'm working on some method to give a field a "no value read until now" or "no initial value received", so that a screen desigend for a specific car could be used for any other too, but the user will "see", that some values are not for him ...

yoh-there, 2015-09-21:
I don't understand. I now check before adding if the field exists in the definition. If not, the listener is simply not added. No exceptions (but I do display a toast, as it usually means I made an error). Just using "a" car type is more prone to having undetected errors I would think.

I do agree on i.e. returning a null value when there is no response.

For now, I think we can close this specific issue?

fesch, 2015-09-21:
yes :-)

closed 2015-09-21 by fesch