fesch/CanZE #24: ELM performance

fesch, 2015-09-20:
I think that some of the ELM commands can be issued only once and do not need to be resend on each request, so putting them in the init method could speed up the polling.

I will investigate on this ...

yoh-there, 2015-09-20:
True if you sort the filters list on network address. I don't think any commands can go to init, but issuing different queries to the same network device you can can just only send the non-at commands one after the other, only doing the filtering stuff once.

For the free frames, I don't think it can be further optimized. It is actually a horrible way of doing things. I now prefer to find all through the ISO-TP methods.

fesch, 2015-09-20:
I've tested this manually the other day:
- atfcsh with any request ID
- atfcsd300000
- atfcsm1
As said in the documentation, the atfcsd-command needs a atfcsh command first, but as long as no reset is done, the atfcsd-command with its data is hold in memory. atfcsm1 indicates the mode and as long as I don't change it, it stays on mode 1. Thus:
INIT: +3 commands
per REQUEST: -4 commands

OK for the free frames, there I found nothing to improve :-(

yoh-there, 2015-09-20:
Wonderful!

And, sorting the filter list on network address will work. As a LOT of info can be asked from the EVC, that could reduce 2 more commands as long as the ECU is the same

Dang, I am having merge conflicts ;-)

fesch, 2015-09-20:
Yes, something like:

if ( the last used requestID != the actual requestID ) then send "atsh ..... "
else do nothing :-)

fesch, 2015-09-24:
Implemented & tested --> it works!

yoh-there, 2015-09-24:
Yay! Should I try that "do nothing if same ID"?

fesch, 2015-09-24:
Of course: That could be another optimization and would make us win at least 100ms per request.

fesch, 2015-09-24:
Actually it would be nothing more that an if around the atsh and atfcsh commands:

if(lastID!=field.getId()) {
    lastID=field.getId();
    ...
}

yoh-there, 2015-09-24:
Yup, but hey, I am still in the Android learning curve :D WIll do that now

yoh-there, 2015-09-29:
While the issue is closed, I think we can improve performance and stability a tad more. Let me know what you think please.

We now almost always wait 50 ms before issuing an AT command, and that is after clearing any outstanding input. This is not needed if we wait not only for an EOM, but actually for the > sign. It literally means: ready and other than just guessing the 50 ms, it is a true sign the ELM has done it's thing.

Furthermore, I am still struggling with the ATMA commands. I believe the timeouts are still wrong. The 400 ms wait needs to be between the ATCRA and the ATMA. Now it is before the ATCRA (not needed), and 80 between ATCRA and ATMA. What I tested back then (see spreadsheet), this was not enough to have a reasonably error free and correctly filtered ATMA output.

Furthermore, I think we need to check responses far more carefully: we now assume the answer is OK, but things can go bezerks quite easily.

I propose we discuss above suggestions thoroughly, reopen this issue when that is done and if we we decide to implement that, we might or might not start by duplicating the current device to something like ELM327bis or something, as to not screw up things too bad while avoiding nasty merges.

fesch, 2015-09-30:
Sorry, but you are wrong. the order is this one:
- empty COM buffer
- emit command
- sleep

Thus ('atcra',400) + ('atma',80) will put a delay of 400 just after the ATCRA command.

OK for checking the OK answer. This will, so I think, not have any impact on the performance, but of course on the stability. Yes.

yoh-there, 2015-10-01:
Sorry for missing your last comment.

I (partly) misread the code. Indeed sleep after emit, thanks for pointing that out. I was thrown off by waiting BEFORE checking an answer. That is relying on buffering during the wait and I now understand better why things sometimes tend to overload.

Anyway, it's a bit futile, as I am implementing waiting for > (per the spec), indicating the ELM is now in rest and checking for OK, indicating things are (mostly) going as intended. From what I tested (laptop, not app), no waiting after receiving an > is never needed, except between cra and ma. That is so specific that I will implement it in a separate method for monitoring.

fesch, 2015-10-02:
I yesterday spend over an hour to test the ELM. I used the very long answer frame of the cell voltages. Generally the values came in quite fast, but sometimes, or even often, thing went horribly wrong. Not only he received values were absolutely nonsense, but also the speed. They dropped in one by one, instead of all nearly together :-/
I suppose some kind of dead lock occurs somewhere, but I didn't find any something until now.
This is definitely a problem of CanZE (I think at least) and not of the ELM itself.
I will have to check everything from bottom up, method by method ... maybe draw a nice schema on how thing interact. There must be somewhere something ...

yoh-there, 2015-10-02:
May I then suggest to at least take a look at ELM327experimental Bob? I
think the two main methods (the query loop and the sendcommand) have become
way too complicated and I have ripped things apart and went back to my
original coding in C, transcoded to Java. It is way easier to add decent
debugging there, IMHO.

I saw another "issue" yesterday. When in the main screen, there is still
traffic between CanZE and the dongle. This is new and unintended. It seems
like the filter listed is not cleared. It was past midnight, so I either
may be wrong, a one-off or intended behaviour. I couldn't find it in the
code.

fesch, 2015-10-02:
Of course I will do that ;-)

Th traffic between the MainActivity and the ELM dongle *IS* intended! This is new, *YES* as it is going to read the speed and block the buttons in case the car starts moving.

yoh-there, 2015-10-02:
Thanks for clearing that up! I noticed it first when an activity crashed and though a thread was orphaned, then rebooted my phone and it happened again without the crash ;-) How to fool yourself!

yoh-there, 2015-10-05:
No time to work on ELM327experimental much. However, I think we can make a huge speed improvement by implementing a dynamic "done" character. Now we simply wait the full timeout, then eat whatever the ELM answered, a huge waste of time: the only place where this is useful is between CRA and MA, and there we can simply issue a Sleep.

What we should do is to also (next to timeout) break the input loop on a programmable character. Usually this is the > (for all at commands except ATMA, ATZ and ATWS), sometimes a <cr> or a ?. I bet this will at least double the speed and it will make things less dependant on the actual speed of the dongle.

If you (Bob) agree, who should implement this? We can determine the stop character either within sendCommandAndWait (based on the command parameter), or pass it as a separate parameter.

fesch, 2015-10-08:
I've did some changes to ELM327experimental. 

In fact it did not display any data, because the data pushed to process was in the wrong format. Also ISO-TP message assembling was missing. Either that, or I am not working on the right version and some push/pull did not work as excepted :-/

The driver now works, but I cannot really notice any speed increase. For me, it is still less reliable as the normal one. In the Battery screen for example, the temperatures need a long time before they are getting displayed/captured.

yoh-there, 2015-10-08:
Oh shoot, now we have a huge merge clash ;-)

Thank you!

Let me answer you on speed when I have done that. There should be a huge
increase as the ELM class waits the full timeout, whether the ELm is
waiting (>) or not, so especially on ISO-TP frames.

yoh-there, 2015-10-08:
Thanks for the corrections! Yeah, had already notice ie the messages not formatted. What I didn't notice and do now is that in a multiframe, you keep the \r's. There I must strongly disagree from a "elegance" perspective. ISO-TP handling should not add anything to the payload, just reassemble the message. I will make some changes there.

yoh-there, 2015-10-13:
I am closing this issue. All improvements have been implemented (though I still don't agree on sleeping the full timeout parameter, it should stop on a stop character too, usually a > or ? sign, sometimes a CR).

The ELM handling is complex and needs baby-steps improvements, which seem to be happening just fine.

closed 2015-10-13 by yoh-there