fesch/CanZE #26: Add a display / clear DTC activity

yoh-there, 2015-09-20:
This would be thrilling, and also negating the need for an i907 a bit.

It is especially important to be able to reset the BCB (for which we have no CANbus ID yet, as it can completely disable the car when it goes in error.

yoh-there, 2015-10-19:
Sequence for querying (at least for the BCB is) (all ISO-TP):
Send 19 02 3B
Receive 59 92 FF [D1 D2 D3 BB] x n
D1 D2 D3 represents the DTC number, BB a bit encoded status, see  http://raviprashanthsn-technical.blogspot.nl/2013/10/fault-code-status-bits-in-automotive.html

Sequence for clearing
Send 14 FF FF FF
Receive 54

And that's it!!!!

You might need to put the car in "accept clear DTC mode": Enter key card -> Gear to D >- Hold start button for approximately 5 seconds until the "Remove card" message appears -> Gear to P.

solmoller, 2015-10-19:
Great stuff!

fesch, 2015-10-19:
Well done :-)

yoh-there, 2015-10-21:
First DTC experiments are being run today. Stay tuned.

yoh-there, 2015-10-25:
My biggest problems at this moment are keeping the BCB awake or ina state to answer queries. When I power up the car, it gives me one semi-proper reply (ISO TP formatting screwed/aborted) , after that, it is dead. I need to check what is going on with the Due logging. Grrrr.

yoh-there, 2015-10-26:
ISO-TP is stable now from the BCB, but it still quits after ONE inquiry, and it will not restart until a full bus power down. This is not what I noticed with Harm's car @ the dealer, so either the CLIP tool was sending more commands on other addresses (very hard to figure out in the enormous data stream), or my car is acting up.

I would appreciate if one or a few of you could do the following experiment in the latest build of Canze (car powered):
- switch to experimental
- start playground (Jeroen)
- from the three-dots menu: ELM test (just let it do it's thing)
- from the three-dots menu: Query BCB (print screen please)
- repeat the Query BCB a few times. Does it go silent on your car too?
- do the same for Query Cluster (instrument panel). The repeat should not die.

yoh-there, 2015-11-02:
DTC querying has now moved to the DTC experimental screen, as it should. It uses standard DTC querying, but I am not too sure all ECU's play nice. I need an i907 again to spy on it. Please giv it a spin.

yoh-there, 2015-11-16:
I am going to close this issue.  DTC handling has moved to "Technical". I am reasonably sure the implementation is not fully correct right now, but I don't have enough cross checked (i9097) confirmed DTC's at hand. Feel free to reopen.

I am specifically including @HarmOtten here, as he has both CanZE and the i907.

closed 2015-11-16 by yoh-there