fesch/CanZE #29: OBDII BT Dongle test

solmoller, 2015-09-22:
We are going to have a lot of people testing this app with various dongles. Apart from a list of known good/bad dongles in the hardware section on the wordpress site, we could add a test to the setup menu. 

I suggest adding a button at the end of the BT setup: 'Test'. Pressing it makes the app issue
ATZ
- and display the result, in my case 'innocar'
Then issue:
ATSH 7E4
- and display both the result and an interpretation : 'OK 5 bytes read - that looks good'
Then issue:
ATFCSH 7E4
- again display the result and an interpretation : 'OK - two times OK, it appears you have a working adapter'

We might go as far as to do some speed testing, but that is outside what I can describe

Henrik

fesch, 2015-09-22:
Good idea :-)

yoh-there, 2015-09-22:
OK, I will take that challenge on.

fesch, 2015-10-12:
The best place to do this is place a button in the Settings menu that is only visible/enabled if ELM327 is selected.

yoh-there, 2015-10-13:
I agree

yoh-there, 2015-11-10:
ELM test is now fully implemented in the Playground (Jeroen). Give it a spin, and I'll move it to the Setting screen later.

yoh-there, 2015-11-12:
After some consideration I think we should leave ELM testing in experimental (or a third category, something like "technical" or "advanced", where also DTC should be moved once we're a bit more sure it works.

I am closing this issue.

closed 2015-11-12 by yoh-there