fesch/CanZE #33: Screen with 96 widgets for cell voltages crashes on low ram tablets

yoh-there, 2015-09-22:
Reported by MH (512MB RAM tablet, not much, but a great tablet for CanZe only, about 50 euro's)

fesch, 2015-09-23:
Yeah ... I mean, of course ... this screen is extremely resource hungry. I am planning to add a widget that supports multiple SID's so I could draw a single chart with all 96 values. That will not use as many resources then ...

yoh-there, 2015-09-23:
If I may put in a suggestion, 96 meters, I don't think it will work from a UI point of view ever. I'd suggest a "heat map" would be better (ie 76 green ones, the lowest 10 red, the highest 10 blue, and when tapping that color display ONE meter (and the average maybe).

fesch, 2015-09-23:
Yes, either that or a simple line diagram, the cells 1 ... 96 on the X, and the voltage on the Y. With colors ... yes, why not.

fesch, 2015-10-12:
I've replaced this with a graph displaying the same information which does not consume as many resources.

closed 2015-10-12 by fesch