fesch/CanZE #36: Too tight "hot" loop in ELM

yoh-there, 2015-09-23:
Not implemented as I wanted to be sure first: shouldn't, in case of a character not being available (closing brace line 347), control be released to the OS, i.e. through

SystemClock.sleep(2); 

I know sleep should at all times be avoided on a UI thread, but this is not running in a UI thread I think.

fesch, 2015-09-23:
http://stackoverflow.com/questions/12290961/what-is-the-difference-of-thread-sleep-and-systemclock-sleep-and-which-one-shou
http://developer.android.com/reference/android/os/SystemClock.html#sleep%28long%29

So I think this will not make any big difference here ...

closed 2015-09-23 by fesch

yoh-there, 2015-09-23:
Maybe you didn't push, but now, in that loop there is no "sleeping" at all? Am I blind? Or are you just saying HOW to sleep doesn't make a difference and you will implement it?

fesch, 2015-09-23:
First of all a wanted to say that the method of sleeping does not make a difference.

Secondly there are at least five closing brace lines and as I was not sure about which one you meant, I preferred that you'll put it yourself ;-)

yoh-there, 2015-09-23:
Got it and will do. Thanks! I am trying to be very careful with your work on ELM really!

fesch, 2015-09-23:
No problem ...