fesch/CanZE #37: Custom hardware

fesch, 2015-09-23:
For my tests with the X10 data I switched back to my Due as it gives me nearly real-time data for the free frames.

I've added this issue - OK, it is not really an issue - to not to forget to continue digging into this direction to.

For certain cases the ELM is fine, but for anything that has to be show in real-time it is of no use.

yoh-there, 2015-09-23:
Completely agree. ELM is fine for diag (ISO-TP frames). For free frames, it is not so good, setting a filter and waiting for each individual frame.

For the screens aimed at the common users, I am focussing on using JUST ISO-TP if possible. For speed, we need raw frames fast!

yoh-there, 2015-09-24:
Bob, see https://github.com/collin80/GVRET/issues/8 If negative, I think we should start over / use your code and move to something that can easily be ported indeed to Teensy 3.0.

My second HC-05 came in today, so as soon as I have some time, I will wire it to my Due hardware.

We should not forget to implement something to initialize / pair the BT module if this needs to be user friendly.

yoh-there, 2015-09-25:
Positive reply from Collin. I think you and we should discuss how to proceed now, to avoid doing double or triple work.

fesch, 2015-10-08:
I've managed to put things together so that my Teensy now does the same (bad) job as my Due.

Unfortunately  replies to ISO-TP requests are not always received. Exactly the same problem as on the Due :-(

I will investigate on that again later on by doing some more tests. It seams to me as if it is a software problem here.

yoh-there, 2015-10-08:
Maybe I said it earlier, if so, I apologize.

If you receive a FIRST frame (1X), you have two options.

1. Reply with 300010. The "10" means 16 ms delay between frames. The ECU
should start sending you all frames and you already know how many, so that
should not be very hard.

2. Reply with 300110, The "01" means send ONE frame (instead of 00=all).
You have to repeat sending those responses now for every reply until the
entire message is in.

fesch, 2015-10-08:
I don't get it :-(

5010: sending: 79B,0221410000000000
5022: received: 7BB,107E61410F3A0F3B
5022: type = 1
5022: messageLength = 126
5022: flow = 79B,3000000000000000

The number in front of each line is millis(), just to make sure that the flow control frame is send within the 100ms. 

I thus send 0221410000000000 to the 0x79B then the reply comes from 0x7BB which is of type. So far so good. I then send 3000000000000000 again to 0x79B and wait for some answer, but none is coming in.

"none" is not quite correct. Sometimes there is an answer, but this happens not that often :-( I must do something wrong or my car does not like to talk to me using ISO-TP frames :-/

What I've tried so far:
- add some delay before the flow-control frame (up to 100ms)
- send 300000, so only 3 bytes
- send 300010
- even send 300110

I only get the type 1 frame :-|

I have reduced my program to only the bare minimum to send and receive those frames, so nothing special anymore. Everything hardcoded.

yoh-there, 2015-10-08:
My first impression is this should really work, so, I need to look harder.
Give me  a few minutes

yoh-there, 2015-10-08:
Your log looks fine to me, and since you are reporting occasional success,
the issue MUST be timing. A few things come to mind, I haven't seen you
test code of course, and of course, some suggestions are silly:

- do you wait for the 10 frame to come in? (yes I am sure you do)
- try a little delay before sending the 30 frame (mêh);
- do you decide to use flow control based on receiving a 10 (FIRST) frame?
Of course you need to avoid sending FLOW frames when there is nothing to
flow;

Frame stuffing should not have any impact, I have actually seen the i907
doing both.

Let me find you a log of a i907 conversation, hold on.

yoh-there, 2015-10-08:
A savycan log

fesch, 2015-10-24:
Just to give a little update:
- the free frames are working very well. The polling is very fast!
- the IS-TP frames mostly work and are a lot faster than on the ELM, but this part of my code is not yet very stable (in my opinion). There are certainly some improvements to make!

fesch, 2015-12-20:
My last comment says it all --> closing

closed 2015-12-20 by fesch