fesch/CanZE #38: LeafSpy clone (BOB)

FabienZE, 2015-09-24:
On my mako (Nexus 4) under Cyanogenmode 12.1, launching LeafSpy clone always crashes
I use a Konnweii 902 ELM327

yoh-there, 2015-09-24:
@FabienZE , can you be a bit more specific?

What version, download (which?) Or the actual development software (through Android Studio)?

When? Startup? When starting one of the activities? If so, which one?

Personally, I have no experience with Android 5 (which is the basis CM12.1), or CM itself. It will be pretty hard to nail this without a developer having access to an equivalent hard- and firmware setup. My advice would be, since you are into developing yourself, to install Android studio and running CanZE from the sources. It's logging will probably help us all along.

Maybe Bob has a different view though.

FabienZE, 2015-09-24:
I currently use the latest weekly build. When I push the LeafSpy button, app crashes. 
I rangement into source files, but didnt see first what is wrong. 
I will try to investigate

yoh-there, 2015-09-24:
I am confused. You speak about LeafSpy. We ARE talking about CanZE no????

FabienZE, 2015-09-24:
OK sorry. I speak about the LeafSpy button in Canze. The one that launches LeafSpyActivity. 
With the first build of Canze, it works perfectly, but it doesnt with the latest weekly build

fesch, 2015-09-24:
"LeafSpy" is the name of a button which has the same data behind it as the one that is being diplayed on my friends LeafSpy display ;-)

yoh-there, 2015-09-24:
Ah OK, so just to keep the subject to the point, CanZE starts, it is the "LeafSpy activity" that crashes. I have changed the subject accordingly. If I am still wrong, let me know.

FabienZE, 2015-09-24:
That's the point ! You are in. 
Sorry, again if I was not clear.

fesch, 2015-09-24:
If somebody would help me getting CyanogenMod on my Asus 1801T, I could test. Unfortunately this device is not in the list of supported devices :-(

yoh-there, 2015-09-24:
I have never run the pre-compiled versions I fully admit. I will try that later on, to check if it's really CM or just something flaky this specific build. The project changes SO fast ;-)

fesch, 2015-09-24:
With the latest build on an out of the box Android, the activity does not crash. I've just verified. Next challenge would be to install CM on some device ;-)

solmoller, 2015-09-24:
I have one, will test

solmoller, 2015-09-27:
I have tested todays build on a CM device.Starts fine, byt selecting settings crashes immediately

I managed to get some debug info out of my phone:

09-27 21:58:48.994: E/AndroidRuntime(12400): 	at lu.fisch.canze.MainActivity.onOptionsItemSelected(MainActivity.java:501)
09-27 21:58:49.004: W/ActivityManager(386):   Force finishing activity lu.fisch.canze/.MainActivity
09-27 21:58:49.514: W/ActivityManager(386): Activity pause timeout for ActivityRecord{426fe5e8 u0 lu.fisch.canze/.MainActivity}
09-27 21:59:00.064: W/ActivityManager(386): Activity destroy timeout for ActivityRecord{426fe5e8 u0 lu.fisch.canze/.MainActivity}
09-27 21:59:01.834: I/ActivityManager(386): Process lu.fisch.canze (pid 12400) has died.
09-27 21:59:01.834: I/WindowState(386): WIN DEATH: Window{42661218 u0 lu.fisch.canze/lu.fisch.canze.MainActivity}

fesch, 2015-09-27:
I suppose the Bluetooth thing is working somewhat different under CM. You may test the latest code from the repository where I changed the behavior of the app a bit when starting the settings activity, but I doubt that these changes will help here ...

FabienZE, 2015-09-28:
With the latest build no more crash!

fesch, 2015-09-28:
Fine :-)

solmoller, 2015-09-29:
Ahem, I still have the issue on my CM 10.1 Android  4.2.2 - crashes on opening settings menu

I'll see what I can do

Fuglemat, 2015-09-30:
Same here, CM 11.2, Android 4.4.4, took a quick look. On opening settings menu, app crashes at MainActivity.java, line 556, NullPointerException on 
     device.setConnectedBluetoothThread(null);   
(because if there's no bluetooth around, because you're lounging in your sofa, far away from the car, device is null.)

yoh-there, 2015-10-02:
Is this still an issue? On my stock S3 the settings screen opens without a problem when there is no Bluetooth connection. If it is, please wait for Sunday's release. After trying that, please do leave a comment here if the issue is still valid or is fixed. Thanks!

solmoller, 2015-10-03:
Just did a rebuild, and the problems seems to be solved for me

yoh-there, 2015-10-03:
Oops, after the last pull (tonight, 20:30-ish), regression Crashes, most on closing the main screen

yoh-there, 2015-10-03:
Seems fixed now, thanks

closed 2015-10-03 by yoh-there