fesch/CanZE #40: Propose to rename and a few modules

yoh-there, 2015-09-26:
activity_test.xml -> activity_test_x10.xml
TestActivity.java -> TestX10Activity.java

To avoid further name clashes, I propose to always include what is tested in the name.

As more developers join in, needing a bit of a playground, I also propose to create a new screen with will contain all test buttons and move those out of the way of the main screen.

fesch, 2015-09-26:
counterproposal #1: distinguish between "official" and "testing" screens
counterproposal #2: "finish" some of the screens and mark them as such, eg. move the finished ones at the top of the welcome screen and the other ones at the bottom

Screen size seams to matter: The driving screen is not very readable on my Nexus 7, not to talk about the 19'' Asus 1801-T :-P

yoh-there, 2015-09-26:
Please do not counterpropose the renaming (I think you're not) :+1: 

As for grouping, My proposal basically IS your number one, moving the testing screens to a searate sub-screen.

If like to go for number 2, please literally draw a horizontal line on the screen to separate the real thing from the crappy experimenting.

As for screen size: I agree, but I need some guidance here on how to autosize more or less (apart from some "real" positioning errors)

fesch, 2015-09-26:
OK, finally after implemented that, I noticed that #1 = #2 anyway. I draw a fat orange line between them now.
Rename: done

closed 2015-09-26 by fesch