fesch/CanZE #42: Build date

solmoller, 2015-09-27:
Well are having people opening bugs with outdated builds.

I suggest we follow http://stackoverflow.com/questions/7607165/how-to-write-build-time-stamp-into-apk

And add a build date to the GUI under settings - this will allow normal people to communicate the version they are having issues with. It should also lead them to update the app before logging issues

The code appears simple:
  try{
     ApplicationInfo ai = getPackageManager().getApplicationInfo(getPackageName(), 0);
     ZipFile zf = new ZipFile(ai.sourceDir);
     ZipEntry ze = zf.getEntry("classes.dex");
     long time = ze.getTime();
     String s = SimpleDateFormat.getInstance().format(new java.util.Date(time));
     zf.close();
  }catch(Exception e){
  }

fesch, 2015-09-27:
Nice idea. 
I've added that ...
[image]

closed 2015-09-27 by fesch

solmoller, 2015-09-27:
36 minutes from suggestion to closure. What is this? apart from brilliant :-)

yoh-there, 2015-09-27:
Hee!!!!