fesch/CanZE #44: Kangoo has 4 battery compartment sensors (like the Fluence)

yoh-there, 2015-09-28:
I will fix that in "charging".

Issue reported by "kaimaik" through PM

yoh-there, 2015-09-28:
This was already fixed in the lastest build (in "charging"). I don't think the "battery_temp" screen (with the widgets) has been made "Fluence or Kangoo aware" (though I might be mistaken!), to I will re-assign this one now to fesch.

fesch, 2015-09-28:
Okay, the last 16 cells are now being hidden when Kangoo or Fluence is selected.

closed 2015-09-28 by fesch

yoh-there, 2015-09-28:
Sorry for being an a**. I don't think this is the proper strategy, The frame length is different (shorter!) too. so you're basically pulling data from a non existing end of the buffer. Unless I am misreading the code of course.

fesch, 2015-09-28:
That's right ... but the data coming from the CAN bus may not have the correct length anyway, for example because of a bad com, so is there is anything that goes wrong here, an exception will be thrown which is getting caught and ignored. So finally this field will not get any updates.

But, OK, for performance reasons these field may simply not be connected. I've added that "if" now too ...