fesch/CanZE #47: Very strange vehicle behaviour

hundenmio, 2015-09-29:
I used CanZE on a 20 km drive today, and had a very strange experience.

I was using cruise control, when suddenly the cruise control disables and the speedometer reads 127 km/h for maybe a second before it returns to actual speed, but without cruise control enabled.

This happened a total of three times, and I noticed the odometer and all driving data had been erased after these incidents.

A few things were different in my set up. The phone was connected to both rlink and elm327 via bluetooth.

I was showing the driving info screen, but had the app in the background as i was using Spotify. 

I *think* it happened when the phone went into standby all three times, or at least a little while after the phone entered standby.

The phone is a LG G4. The ELM is a ver 1.4 knock off China adapter. Android 5.1.

solmoller, 2015-09-29:
What screen were you on? We should have prevented standby by now

I've tested for hours with an ELM with no such behaviour, but we should
take this seriously

Henrik

hundenmio, 2015-09-29:
I was on driving parameters, but had Spotify in the foreground.

yoh-there, 2015-09-29:
Dear @hundenmio ,

Thank you for the details (but we need some more), and yes, this sounds a bit scary.

Can you please provide us with the following information:
- what build were you using (visible in the setting menu)
- tell us a bit more about the dongle please:
  - did it provide reasonable information in the app?
  - is a KonnWei or MaxiScan?
  - can you tell us what you payed (it is an indication about certain features, or lack thereoff)
- what car, and did you set it in the settings screen?

Thing is: we do not write anything on the CANbus, except standard diagnose messages, which the dongle should format. In all seriousness: unless there is more diagnosis about what is going on, I would not feel confident about using that thing. Not putting any blame, it might be the app indeed, but as Henrik said, we both have driven a lot with it running. There are stability issues for sure, but not fooling the car.

hundenmio, 2015-09-29:
Using latest build. 

The dongle, not sure what I paid. Have used it for quite some years. I'd say it's your standard run of the mill knock off no name china ELM327 dongle.

No info in app as far as I know. As the phone went into standby, I'm not really sure.

Cat is Zoe 21 model. Set up in menu for sure.

Not saying the app is to blame, but it is a really interresting bug somewhere. For all I know it is a coincidence and nothing to do with either app or dongle. I've used the first three public builds a whole lot myself, never seen this before today.

hundenmio, 2015-09-29:
It happened two more times. The app being in the background is not a contributing factor. Will happen as long as the driving parameters screen is selected.

Also, I was mistaken about the cruise control. It does indeed reactivate after the speedo has shown 127 kmh. It seems like the car somegow thinks it is travelling at that speed for a second. 

Not sure, but the real speed display in the app showed higher speed than I was travelling at the two last times. Not 127 kmh though.

Could not replicate the issue in other screens or without the dongle.

yoh-there, 2015-09-29:
See the manual about the speed. it is UNCORRECTED speed. So by design.

hundenmio, 2015-09-30:
Yes, I know it's uncorrected speed. The app displayed higher speed than the cars speedometer when the problem occured.

yoh-there, 2015-09-30:
Speed: gotcha, I misread. Thanks for pointing that out.

I suspect timing issues, but I can't be sure. These things are VERY hard to debug, let alone without the culprit at hand. My humble suggestion is to start with buying a KW902. The knockoffs are impossible to identify and all their firmware is stolen and misreporting versions. Given that we do not (intent to) send anything on the bus, it is extremely worrying that the dongle DOES send something that pushes the car over so to speak. There is a reason why we "support" these only.

For now, I am going to leave this issue open as there are things to do with the ELM driver, and I will assign it to myself. Reason is it might encourage others to report similar issues and we need to know if it indeed is an app or a dongle issue.

fesch, 2015-09-30:
I agree: we do not want to "interfer" in any way with the normal operations of the car. I personnaly think
such would be too risky, to dangerous and I even suppose that would not be too legal either ...

hundenmio, 2015-09-30:
I'll consider trying a KW902 instead. It might even be a little faster.
Did the same run as last night this morning, this time without the dongle. Everything was normal. 

Is there anything I can do to assist you with this matter? Any logs or logging I could do?

yoh-there, 2015-09-30:
Thank you for your offer, we'll probably take you up on that one as we feel it is very strange indeed. I need to get back to the others if we can build in some easily accessible logging. I will post that here, so you should get an email if this issues is updated.

yoh-there, 2015-09-30:
In addition, a few days back I changed getting the speed from actively querying an ECU to sniffing on the bus. Please rerun your test after next Sunday's weekly build with your dongle, thanks.

hundenmio, 2015-09-30:
Will do.

yoh-there, 2015-10-13:
As we didn't have any more reports on this, I am closing this issue. Feel free to re-open or re-issue. Personally, I suspect a not-so-compatible dongle, so I would appreciate an as-accurate-as-possible description of it (link if still on sale, picture, version string as reported by ATZ).

As for the legal implications of interfering, I think for now we have tackled that sufficiently.

closed 2015-10-13 by yoh-there