fesch/CanZE #50: 2015.09.27 build: Not connecting and crash on opening settings

ahetlelid, 2015-09-30:
The latest build does not work on my LG Optimus 2X stock rom. There is no connection established and when I enter settings the app crashes.

Running the 2015.09.20 build, settings and connection works. The Leafspy page crashes in this version, but the charging page works just fine.

yoh-there, 2015-09-30:
@ahetlelid   What Android please?

Team: maybe this is a duplicate of #38 and it is unrelated to CM??

ahetlelid, 2015-09-30:
Sorry. Forgot to check that information. It is Android 4.X, but I am unsure about the exact version as this phone has not been updated by LG for a loong time. (I am currently only using it for occational android necessities). It was updated last time when LG released the 4.X stock rom.

I will check when I have it available again.

Fuglemat, 2015-09-30:
Sounds like the same problem, anyway, I did this

                    // stop the BT device
                    if(device!=null) {
                        device.setConnectedBluetoothThread(null);
                        BluetoothManager.getInstance().disconnect();
                    }

and went out and tested it, and it works. LeafSpy also works.

yoh-there, 2015-09-30:
@Fuglemat Did you make a fork? Are you pushing your changes to it and can Bob merge those? If needed I will paste here a simpe (and not very state of the the art) step by step on how to collaborate on github with Android Studio.

fesch, 2015-09-30:
Added manually that if ;-)

closed 2015-09-30 by fesch

Fuglemat, 2015-09-30:
I just cloned it to take a quick look, and I had to disable all Git stuff to make Android Studio run on my old laptop. I would suggest you just fix it yourself for now :)