fesch/CanZE #52: Add software version screen

yoh-there, 2015-09-30:
Rationale: On the fora, often there is a mention of a software update of a module (recently the BCB and the airco) and dealers seem to not always install the latest stuff. The impression I get is they get an advisory when they report in your car at Renault but they don't always follow it. After all, it's an advisory.

Alexandre supplied me with the proper commands. I will check them out and implement those if they work, preferably for all 16 (for the Zoe) ECU's. I would appreciate if a Fluence owner could check if all 16 are present. See the spreadsheet for all the ECU adresses.

yoh-there, 2015-10-02:
Build the sreen. Guys, could you please do me a huge favor? It doesn't work, and I assure you, the principle does work (I ran it in my C code and it is just fine). I simply "don't see it". Feel rather daft to be honest.

yoh-there, 2015-10-02:
I think I got it, inspired by issue #57. And I think it also made me find the major instability. If an ISOTP query is send to a non-existing address, things go awry it seems!! Finally a clue :-)

avataranedotas, 2015-10-02:
On the Fluence the following ECU's don't respond:
76D/74D
7BC/79C
793/792
722/702

yoh-there, 2015-10-02:
Thanks, we can exclude them later using the car setting. At this moment I am trying to make the ELM driver not getting into lethargy when there is no responding ECU.

In the zoe the following ECU's do not respond:
76E/74E
793/792
722/702

yoh-there, 2015-10-02:
Done!

closed 2015-10-02 by yoh-there

yoh-there, 2015-10-03:
Added reference version in the code, per car (not per hardware variants, we need to figure how this pans out).

I would appreciate if you all post your results, so I can update the "golden standard" reference. I have ECU arrays for Zoe, Fluence (empty), Kangoo (empty) and X10 (copy of Zoe)