fesch/CanZE #64: Load firmeware versions from external online source

fesch, 2015-10-03:
The idea is to move the versions from the code and
- load from from the local preference file
- add a call to a server (canze.fisch.lu) to allow dynamic update from the web

yoh-there, 2015-10-03:
You mean the field definitions? Cool idea

Unrelated: we could push software versions to a web server to automatically record the highest seen, but I don't like spyware

fesch, 2015-10-03:
That's a cool idea too, but in that case I'll add two buttons
- one for "update latest seen firmware versions"
- and another one fir "send my firmware versions to server" (with all necessary security related checks)

To load the field definitions from the web is a good idea too, but for later on I think. Once the software does not change as often as now, there is no need for this ...

yoh-there, 2015-10-03:
Ah, I get it. I misunderstood your original idea. Yes, great plan indeed. I would propose to activate it through the menu (top bar) of the firmware screen, not on the screen itself, main, or settings screen.

For uploads, https required, and a SHA256 hash over the payload, including a secret key?

fesch, 2015-10-03:
Https is maybe not needed, but a secret key will. We need to make sure nobody will try to send faked data.

yoh-there, 2015-10-03:
Stubborn man, hehe ;-) The key is needed to prevent fake uploading. https to prevent eavesdropping

yoh-there, 2015-10-03:
BTW, if we do this, we need to design this in a way that we can include the roughly 5 other "version" fields per ECU. I don't fetch / display those right now, but will probably do that later in a popup-per-ECU line. Just saying it would be smart to save a reference version for those fields too, just in case.

fesch, 2015-10-04:
I would like to use JSON, as this is easy to generate, easy to read and easy to parse. Going this way, we could generate everything dynamically:

[
	// CAR_FLUENCE
	[
		...
	],
	// CAR_ZOE
	[
		{
			'request':'7ec',
			'fields' : {
				'128':0x0203
			}
		},	
		{
			'request':'7da',
			'fields' : {
				'128':0x0112
			}
		},		
		{
			'request':'7bb',
			'fields' : {
				'128':0x0651
			}
		}		
		...
	],
	...
]

The request is the first part of the SID to ask for. The "field" field contains the bits to query with the related max value seen so far.

yoh-there, 2015-10-04:
Fully agree on json. Let me think a bit on the definition please.

yoh-there, 2015-11-16:
I propose to close this this issue with a "wontfix" label.  It just doesn't seem worth it without also sending back data, which would open up a huge can of worms in terms of data protection, who would host it where, etc.

fesch, 2015-11-16:
I agree.

closed 2015-11-16 by fesch