fesch/CanZE #66: Hard reset failed

solmoller, 2015-10-04:
When the app runs in the background it keeps trying to communicate with the OBD - fine, but it should stop complaining about the missing connection on top of other apps.

fesch, 2015-10-04:
Heu?
On my device, the BT is stopped as soon as the app is no longer in background. The communication will be shut down cleanly to make sure the ELM is no in an inconsistent state.

solmoller, 2015-10-04:
I have the app still running in the background, every five seconds I get the message

fesch, 2015-10-04:
What message do you get?

solmoller, 2015-10-04:
I get this: "Hard reset failed :-("

yoh-there, 2015-10-04:
Which is what I programmed in ELM after it detects a problem, tries a soft
reset, fails, tries a hard reset, fails again.

solmoller, 2015-10-04:
.. And continues to warn about it despite the app being in the background. Apps in background should not pester their users.

yoh-there, 2015-10-04:
Yes they should if things go wrong. However, it is debatable if it makes
any sense to run a "display" app when it is not in the foreground. IMHO.

fesch, 2015-10-05:
When riding to school this morning:
- opened CanZE
- checked the Battery screen
- opened Driving screen

After 2 or 3 km, the message "had reset failed" popped up. 
After restarting CanZE (I did not touch the dongle), it worked again nicely until the end of my trip.

yoh-there, 2015-10-05:
So, we need to do even harder resetting. Hmmmm.

yoh-there, 2015-10-13:
There have been numerous changes to the ELM driver and when to run and not run the query thread. that I am closing this issue, but feel free to re-issue-or reopen.

closed 2015-10-13 by yoh-there