fesch/CanZE #80: 2015-10-05 build does not work at all

HarmOtten, 2015-10-06:
The application starts, but does not connect to my Konnwei. When going into the setting screen, the app crashes.

fesch, 2015-10-06:
Did you pair your device with the Konnwei before starting CanZE?

HarmOtten, 2015-10-06:
Yes, it is paired. When I go back to an older version of CanZE, it still works.

yoh-there, 2015-10-06:
Even if not paired at all (maybe Harm has an empty bluetooth paired list?, can you report that please Harm) the app should not crash, so I am labeling this as a bug. As I fiddled with the device screen I will assign it to myself (assuming for now that is is root).

yoh-there, 2015-10-06:
Ok, so an empty list is not the problem (and looking at the code should be no problem if it was empty either). So, already I am giving up, sorry.

fesch, 2015-10-06:
Harm, do you know what version exactly was working?

HarmOtten, 2015-10-06:
Version 2015-09-20 was working, and also a version I got directly from Jeroen (2015-10-03).
FYI: I run it on a astar tablet, with Android 4.4.2.

fesch, 2015-10-12:
Should be fixed with today's build ...

closed 2015-10-12 by fesch