fesch/CanZE #84: Save KM to dest when switching screens

yoh-there, 2015-10-07:
By mail, MH

@fesch : I get no grip on wat is saved between screens in Android and what not. Can you enlighten me a bit?

yoh-there, 2015-10-07:
Is a decent strategy to save in a public property of MainActivity? Or is that totally destroyed too when switching to another activity? I realize IF it would work, it would not survive exiting the app.

fesch, 2015-10-07:
You may push it into a public of even static attribute of the MainActivity. That would work when shitching between the scnreens, but yes, the information will be lost if the app is closed.

Better use a sharedPreferences file. Take a look at SettingsAcitivity, line 45

yoh-there, 2015-10-07:
Yep, I get that. Would you like me to use the same SharedPreferences blob for this (lu.fisch.canze.settings) and commit i.e. a kmToDest variable to it (I assume just doing that would not kill the existing variables in that blob), or should I instead always create a new SharedPreferences blob per activity? What is the "best practice" here?

fesch, 2015-10-07:
I have no ieade what is the best practice here, but I suggest to use the SAME file, as it is the same application. Otherwise the filesystem will be populated with too many files I suppose.

yoh-there, 2015-10-16:
I fixed this, improved the logic, and added "double destination distance (for round trip usage)

closed 2015-10-16 by yoh-there