fesch/CanZE #85: could the font size be enlarged for tablet use?

yoh-there, 2015-10-07:
By mail, MH

@fesch : How to approach this (we are obviously talking text oriented screens)? New layouts per device seems a bit overdone. Can the fonts be made relative to the screen size?

MH suggests a setting but I think that is "mêh"

fesch, 2015-10-07:
We should start using the "sp" unit

https://www.google.com/design/spec/style/typography.html#tall-scripts

yoh-there, 2015-10-07:
I tried in one label. Didn't change much when I switched the emulator to a
big screen.

Is there a way to default this? I must be overlooking something, can't seem
to put that in the overal LayoutRelative or something?

MartijnEV, 2015-10-07:
What about using a 'responsive' UI design (responsive CSS)

yoh-there, 2015-10-07:
@fesch is more experience then I am, but the answer is yes/no. Android apps do not use HTML/CSS for styling. However, all Layouts can be made in multiple versions for different device layouts and I assume portrait / landscape orientation too.

What would be nice if we could have a team-member who would be able to do this sort of work and pickup where "programming" stops. A tiny bit of style policing wouldn't hurt either. All current team members except @fesch are climbing the steep Android developing curve and the technical nitty gritty is already pretty overwhelming (at least for this old dog). Adding that to the coders workload would grind things more to a halt than we'd like.

IMHO.

fesch, 2015-10-08:
I found some interesting and quite manageable way here, but I did not yet test it:

http://stackoverflow.com/questions/6927441/different-text-size-for-different-hardware

fesch, 2015-12-20:
I'm using a 7'' tablet. Font size is OK for now. If someone asks for something more specific, we could investigate more. Until then I close this issue ...

closed 2015-12-20 by fesch