fesch/CanZE #87: Fans on/off indicators

MartijnEV, 2015-10-07:
Is it possible to show an on/off indicator for the
- front / BCB fans (easy to hear)
- rear battery fans (soft, not so easy to hear, only on when really hot)

Just a nice to have

yoh-there, 2015-10-07:
I think that would be a great idea. @MartijnEV could you browse the documentation please for field candicates?

Rumour is that Zoe can also cool the battery pack using it's airco. Does anyone know if that is true?

yoh-there, 2015-10-09:
I am marking #92 as a duplicate: any fan screen should include some key temperatures.

While the original is older, this item has a better title description.

@MartijnEV , any progress?

yoh-there, 2015-10-17:
@MartijnEV Have you found the fields you are interested in in the documentation?

MartijnEV, 2015-10-18:
@yoh-there  I have not yet found the specific fields that we could use. I see multiple messages related to 'clima' etc. but the battery fans are or are not fully related to the clima. So I need to dive a little more deeper into the documentation.

yoh-there, 2015-11-16:
I am closing this issue for now. I think it needs to be combined with #105 anyway, but so far there seem to be not much enthusiasm to really put effort in building it. Feel free to reopen.

closed 2015-11-16 by yoh-there