fesch/CanZE #88: Control Pilot time plot chart

MartijnEV, 2015-10-07:
In the Charge section it would be very nice if we could have a time plotted chart of the control pilot value, which would be plotted during a charge session, (and ideally also be saved until the next charge session starts).

This way we can have a "peek" into what is happening during a the smart charge process, or load balancing process, if present on a particular charger.

yoh-there, 2015-10-21:
@fesch Being the widget man, would it be something you're interested in to add to i.e. the Battery activity?

fesch, 2015-10-21:
Yes, I'll do that. I'm working anyway yet on a method to make the graph data persistent.

I am myself annoyed by the fact that if I change between the screens, the plotted data is lost :-/ Certainly the time when the requested fields are not being polled will miss, but at least the previous data should remain.

For the charging, I will have to add some "stop" event. Indeed if you start charging in the evening and come back in the morning, the graph may be moved to far so you won't see nothing anymore. Or maybe change the timeplot. I'll see ...

yoh-there, 2015-10-21:
I do know you mean adding logic to stop the sampling, but at least here is your trigger:

public static final String SID_ChargingStatusDisplay            = "65b.41";

3 is charging, anything else is not charging (waiting, unplugged, etc)

fesch, 2015-11-04:
I'm working on something ...

yoh-there, 2015-11-16:
I think now that using the control pilot is not the proper field to use, as we don't know the the phase count. Better would be to use AvChargingPower (427.40)

yoh-there, 2015-12-21:
This issue is not forgotton. I am working on some graphs here. Should make it to the experimental section any day now. BTW, the widgets are not very easy to use, an d I really love the gradients.

yoh-there, 2015-12-21:
Done, needs some debugging, but closing

closed 2015-12-21 by yoh-there