fesch/CanZE #89: CanZE crashes while driving in Breaking screen

fesch, 2015-10-07:
This happened today twice within some minutes of the start :-/

yoh-there, 2015-10-07:
Noticed it too as being the least stable screen. I will look into it, but wouldn't mind an extra pair of eyes looking for obvious stupidities.

yoh-there, 2015-10-10:
I don't see anything wrong in the code. The only thing different from at lest the screens I wrote, is that it is the only one doing ATMA queries (meaning a SID with only one dot) exclusively.

yoh-there, 2015-10-12:
I added a useless ISO-TP command to it, to check if it improves stability.

yoh-there, 2015-10-16:
I am going to close this issue. It is either the addition of an ISO-TP command or the improvement of the ELM driver (which is what I hope). I will test privately with removing the ISO-TP again. 

Let me know if the problem still occurs please.

closed 2015-10-16 by yoh-there