fesch/CanZE #91: Copied from spreadsheet: Battery balancing

yoh-there, 2015-10-07:
By AM

yoh-there, 2015-11-12:
@avataranedotas Any progress here Alexandre? I can only say that for the Zoe it is, as far as I can see, not equal to what you suspected for the Fluence.

avataranedotas, 2015-11-13:
In the Fluence it seems to be exactly as in the Nissan Leaf.

Requesting 02 21 07 on the LBC. The 4 least significant bits on each byte will represent the active cell shunts. On the first byte, bit 3=cell 1, bit 2 = cell 2, bit 1 = cell 3 and bit 0 = cell 4. Second byte cells 5 to 8, so on until you get the 96 cells.

yoh-there, 2015-11-13:
OK, what I will do then is try that command one more time on the Zoe, report back here and in the spreadsheet and then close this item. Thanks!

yoh-there, 2016-12-14:
I will close this issue soon, as we have managed to get definition of diagnostics fields. I will change the implementation of this issue to "balancing on/off indication on the advanced charging screen".

yoh-there, 2016-12-18:
Closed, implemented in commit #362

closed 2016-12-18 by yoh-there