fesch/CanZE #93: Copied from spreadsheet: Number of charges slow and quick

yoh-there, 2015-10-07:
by AM: Nissan Leaf records the number of times it has charged, level2 and quick

yoh-there, 2015-10-09:
To figure this out, either:

1. the car would need to keep those counters internally (I suggest you take a very good look at the material we have at our disposal now), or.....

2. We need to implement this through dedicated, always on hardware (arduino), or......

3. The app should always be running, which seems unrealistic.

I propose that @avataranedotas checks for 1., if found, report it here and I will add it to the charging screen, or close the issue.

avataranedotas, 2015-10-09:
I'm quite sure the number 1 is the correct one, as the renault dealers get those values through telemetry.
I'll try to figure out where to read the values.

yoh-there, 2015-11-02:
I have not found the number in the free frame documentation at all. Of course it might be buried in the diagnostic command set of the BCB (or PEB or LBC), but maybe Renault is simply counting? Relying on each session at least charging session dropping one log?

fesch, 2015-12-20:
No progress for > 2 month and no sign at all that these frames could be found --> closing

closed 2015-12-20 by fesch