fesch/CanZE #98: Astar tablet not recognized in Studio

yoh-there, 2015-10-10:
The popular 7" astar tablet (40-ish euro's for a 7 inch display) is not playing nice with Android Studio. On a Windows PC it failed to install drivers when connected, so Studio didn't notice it being connected at all. On Ubuntu it was seen but despite it running stock Android 4.4.2., Studio reported it running API level 1, and thus refused to give it a go.

This is not a CanZe issue of course, but if somebody knows the solution, it would be nice.

fesch, 2015-10-12:
Had the same today with my Terra Pad 701. Needed to restart the Studio while my device was connected to make it work ...

HarmOtten, 2015-10-13:
Unfortunatelly, that did not work in my case. The problem is probably a bit more low-level. First Windows must recognize the tablet. I asked the seller of the table for a driver today. No response yet.

yoh-there, 2015-10-13:
Maybe this helps

https://play.google.com/store/apps/details?id=com.hexamob.drivers

HarmOtten, 2015-10-13:
There is a log list of drivers in this App, but astar is missing. Unless someone knows which brand might be compatible with astar... then I could manually select it.

yoh-there, 2015-10-14:
Anyway, while it would be great to find a solution here, and so we should move it to the developers WiKi here when it is fixed, the issue at hand is totally beyond our control. Therefore I will close it as "won't fix". Sadly.

closed 2015-10-14 by yoh-there