fesch/CanZE #99: ELM initialization

avataranedotas, 2015-10-11:
The ELM initialization sequence starts (toughness = 0) by checking the response to the atz comand (reset). If the response is blank the initELM will return false forcing a retry.

My ELM (KW902) seems to remember the ATE0 command (no echo) after disconnecting from the app, so on the next connection there will be no response to the atz command.

Wouldn't it be better to turn echo on before sending the atz command and then back to off after checking the response?

yoh-there, 2015-10-11:
That is VERY strange behaviour as ATZ should be 100% the same as power off/power on (ATWS is the same, just no LED flashing, so faster). An ATZ should always report back the version "wakeup" message.

I will go out and try this immediately. Can you please provide me with a detailed description what you did, so also please with what programs you checked? For now I assume.......

- any terminal emulation
- ATZ
- reponse ATZ (the echo of the command) ELM xxx version xxxx >
- ATE0
- response either ATE0 OK> or OK>
- ATZ
- response >

correct?

avataranedotas, 2015-10-11:
You are right, my fault!
The problem is elsewhere.

Here is what happens: (Sony Xperia M2 and KW902 dongle)
When opening an activity (charging for example) I get the following "toasts"
"No answer from ELM ... retrying ..."
then right after
"ELM is now ready ..."

When I rotate the screen I get
"Hard reset failed :-("
then
"No answer from ELM ... retrying ..."
then right after
"ELM is now ready ..."

When exiting the app I get the "Hard reset failed :-(" toast

yoh-there, 2015-10-11:
I have noticed that too. I am not sure yet where that problem is. A dongle reset is requested whenever something is "not right", i.e. a timeout or an out of sequence frame from multiframes. I don't understand why/how a reset could be requested on app closure.

Just for the record, are we talking current development version? And ELM or ELMexperimental?

avataranedotas, 2015-10-11:
Latest commit, ELM driver.

A correction to my previous description, the "toasts" appear when rotating the screen only in the main activity, if inside an activity the connection holds and no toasts appear.

yoh-there, 2015-10-11:
Loud and clear. It seems like the communications with the device is not aborted gracefully with things like onPause and onDestroy. I believe the same issue happens when closing the App; sometimes it leaves the ELM in the middle of an ATMA command, and unfortunately, the dongle doesn't even care if the Bluetooth link (blue LED) is closed.

So, I think we need to implement something like:
- when onPause and/or onDestroy, notify the running thread doing the queries that it should stop;
- not kill that thread, but let it kill itself, finishing what it is doing. It should also call the restoreOrder function just before it dies
- avoid problems like removing a query while the driver query loop is in the middle of running it.

When I was young (that is a LONG time ago) I was taught to implement behaviour like this with semaphores. Unfortunately, I am not smart enough to do that on Android. Yet? ;-)

Mildly related: the restoreOrder method in ELMexperimental is very reliable. Maybe it toasts too much, but it has reliable code to bring the device in a known state from pretty much any unknown state.

solmoller, 2015-10-11:
Semaphores? That takes me back to 1995ish...

Btw I'm not participating much presently.  Too much work, and now vacation
away from the fluence. Actually I am sitting in a field awaiting 'black
sun', 100k to a million starlings should gather here in five minutes if
they have set their clocks properly.

Have a great week

Henrik

fesch, 2015-10-11:
Semaphore is replaced by a block called "synchronized" in Java.
I'll take a look at it when the children are in bed.
Just cam back right now from an excursion and they need me now ...

fesch, 2015-10-11:
Here's a resume what happens when onPause:
- filtered fields are cleared for the actual device
- the actual device's connectedBluetoothThread is being nulled out. This will do a proper stop of the polling thread!
- active waiting for the device to finish
- the BluetoothManager is being disconnected.

This is true for the MainActivity and while writing these lines, I've realized that any other activity does not so :-(

yoh-there, 2015-10-11:
Hi Bob,

Then why does a reset is issued and fails?

As for the others, can you point me to what needs to be changed, thanks!

fesch, 2015-10-11:
Rotating destroys the top activity an re-created it.

I will make a commit very soon ... just one thing that does not yet work as expected ...

fesch, 2015-10-11:
The BT is now being shut down if any of the activity that inherits from CanzeActivity is paused (put to background), rotated or destroyed. The restore also works fine. At least during my tests ...

avataranedotas, 2015-10-11:
Another question about the ELM327 initialization:
The last 2 commands send are "atfcsd300000" and "atfcsm1" (line 195 in ELM327.java)

My KW902 rejects the "atfcsm1". 
According to ELM327 manuals the flow control mode 1 requires that you also define a header with command "ATFCSH hhh"

Flow control mode 2 only uses the user-defined data bytes in the response, shouldn't this be the choice by using "atfcsm2" instead of "atfcsm1" ?

yoh-there, 2015-10-11:
The fcsh is in the init code?

In any way it is kinda "fake", as it's only purpose is to not having to send the fcsh command command again, the dongle remembers the header.

You can experiment using mode 2, but since the ZE cars do not use a standard offset of 8 between the ECU's "in" and "out" address, I doubt that it would work. But if it does, great! one command less to send over!

avataranedotas, 2015-10-11:
Sorry for deleting my reply, I tested and realized you are right then deleted my question.

I now have gone through the ELM327 init sequence step by step, the only thing I would change is the atz wait time from 1000 to 1500ms, testing with a terminal gives about 1 second between sending atz and getting the ELM327 v1.5 message, so I think it should have a little more margin.

Thanks for the quick replies!

yoh-there, 2015-10-11:
@Bob, I am sorry, I am really trying to understand, but somehow.......

If an activity inheriting from CanzeActivity pauzes, first device.setConnectedBluetoothThread(null); is called. It will null the devices (this) connectedBluetoothThread. Are you saying this doesn't kill the polling thread (it is nicely threading along), but nulls the variable?

If so, the loop will stop after finishing a query, however, within that query, suppose an atma was just started. line comes in, and the process wants to do a sendAndWaitForAnswer("atar", 0); As the thread reference is gone, everything stops, and and in this case, the atma will keep running, possibly overwhelming the dongle beyond reset-ability?

Then again, if connectedBluetoothThread is null, there is no way to tell the BT thread anything at all anymore, right? So, I still think we are not done in the thread business, but I could very well be misreading the code. Wouldn't be the first time ;-)

I think when the querying needs t stop, the running query should really be able to finish.

fesch, 2015-10-12:
Shit, that's stupid :-/ it's like in the good old cartoons where they just cut of the limb they are sitting on. You are absolutely right.

Fixing that won't be that difficult....

yoh-there, 2015-10-12:
@avataranedotas could you please check if you have a substantial difference in reset time between ATZ and ATWS? They are supposed to behave exactly the same, but WS saved a few hunderd milliseconds are there should be no LED flashing.

If so, maybe you can suggest a shorter time for ATWS. Thanks!

avataranedotas, 2015-10-12:
Sending ATWS I get a instant response, equal to any other command, not measurable without making some software to check the response time. My guess it that 200ms is more than enough for ATWS.

Sending ATZ I get 1 or 2 blank lines instantly then about a second elapses till I get the "ELM v1.5"  line.

yoh-there, 2015-10-12:
Thanks, will change that. If everything behaves according to spec, we should never need an ATZ, who cares about the LED test.

BTW, bloody irritating that the device does not send a > after a ATZ or ATWS. Sigh!

avataranedotas, 2015-10-12:
Well, updated project from latest commit (commit 46ccf3ed49a4f903073ea7338140d8039347b722) and built the apk. Locked my Sony Xperia M2 for several minutes after connecting to the ELM.
I think this issue (initialization) can be closed, other problems can be reported on other issues.

yoh-there, 2015-10-14:
I disagree. I have a WIP that uses basically the ELMexperimental's initialization code. As that one checks on replies, I notice a lot, and I mean a LOT of failures when re-initializing. I want to get to the bottom of this, as simply blurping commands might look fine, but messes up in the long run.

yoh-there, 2015-10-16:
I have uploaded a major change in ELM. Drove with it today and I think it has improved quite a bit.

Note that I have abandoned ATD as command (it is there but never called), abandoned ATZ (ATWS is equivalent and faster), added "stop current operation" logic, retry every single command 2 times and the entire sequence 1 time.

I hope it works for all of you too. Seems my older 1.4 dongle is less happy (it used to work), I might fix that later. Will keep this open for a bit, but if no serious complains for a few days, I will close this.

avataranedotas, 2015-10-17:
I've tested the latest commit with the Fluence and KW902 and the ELM experimental definitely is faster to initialize and gives fewer errors.

solmoller, 2015-10-18:
Confirm, that it appears faster now, and stable

yoh-there, 2015-10-18:
Did some minor tweaks and it seems to be fast and pretty stable now. #112 Closing this issue.

closed 2015-10-18 by yoh-there

yoh-there, 2015-11-08:
I have added another minor tweak in ELM that @avataranedotas suggested: no atar needed until an ISO-TP request is started. Another few ms shaved off :-)