fesch/CanZE #102: Distribution not working

yoh-there, 2015-10-13:
Got this on the German forum

=====
I just compiled the source on Ubuntu 14.04. Compiles straight through. Lint complaines about 150 issues, but the apk's are build.

I can't install the release apk (Parsing error), but the debug apk installes and ... runs well on my Cyanogenmod 5.1.1 :D
=====

I am wondering why there sees an issue with the apk's? We never see these when running from Studio? Mime type seems unlikely? Confused.

fesch, 2015-10-13:
Parsing error while installing an APK file is an indication that the APK has been compiled for a higher API level than the one present on the device.

yoh-there, 2015-10-13:
Looking at build.gradle it seems we are building for API 22, which seems ridiculously high, especially since my S3 is Android 4.4.2. Would think API 19 or 20 should be the target , or maybe even 14. I don't think we are doing very advanced Android things?

Anyway, I can't believe the 22, as it wouldn't run on my device? What is wrong here?

fesch, 2015-10-13:
Attention: build API != target API
build API = the API that is used to build the app
target API = lowest API that is targeted by the build an on which the app runs

yoh-there, 2015-10-13:
Oh I see, and that is 15 if I am reading that correctly, so that should be fine. Still a mystery then why the precompiled version would not run on a recent Cyanogenmod. Seems like Cyanogenmod is giving us headaches here and there.

yoh-there, 2015-10-13:
Maybe adding the uses-sdk tag in the manifest would help? Dunno really

yoh-there, 2015-10-14:
This was reported of the forum. Closing this issue

%%
Just checked on your 2015.13.10 version. It also installs and runs well on Cyanogenmod 5.1.1 :-)
%%

closed 2015-10-14 by yoh-there