fesch/CanZE #105: Temperatur Kangoo und Fluence

kaimaik, 2015-10-16:
Hallo,
kann man die Temperatur vom Kühlwasser (Ladegerät)
und die Temperatur der Heizung (Wasser) mit einbauen ?

yoh-there, 2015-10-16:
@avataranedotas: maybe you can have a look at this? I have no idea where to look

yoh-there, 2015-10-16:
Comment in email: 

Beim ICARsoft werden die Temperaturen angezeigt.

yoh-there, 2015-10-16:
That would certainly help. Please state not only the value but also in which module (ECU) you find it and the id of the query. Thanks!

kaimaik, 2015-10-16:
ok.werde ich machen.
Allerdings erst morgen :-)

avataranedotas, 2015-10-16:
In english?

yoh-there, 2015-10-16:
Sorry. "Can you get to the temperature of the cooling water of the charger and water temperature of the heater?" It should be in the i907, so I suspect ISO-TP commands. Have you figured those? We're talking Kangoo, it seems closer to a Fluence than a Zoe, not sure.

kaimaik, 2015-10-17:
ok.

Antworten kann ich nur in deutsch.

ok ?

fesch, 2015-10-17:
Ja, ich kann ja übersetzen ;-)

kaimaik, 2015-10-17:
ich hab hier mal die Werte vom i907 und dem Kangoo ermittelt.

Alles Werte vom EVC

109 = Temperatur der Antriebsbatteriewanne
116 = Temperatur vom Kühlmittel der Heizung
183 = Aussentemperatur HEX Code= 33B1
195 = Temperatur vom Kühler (Ladegerät) mit HEX Code= 33F2
196 = Temperatur vom Ladegerät mit Hex Code = 33F6 wobei
dieser Wert während der Fahrt sehr unterschiedlich ist.

yoh-there, 2015-10-17:
109 = Battery compartment temp. (22 20 01) ?
116 = cooling fluid temp (22 33 7)?
183 = outside temp (22 33 B1)!
195 = cooler temperature from charger ( 22 33 F2)!
196 = ?? temp, rapidly changing (22 33 F6)
These are all ISO-TP queries directed to the EVC

avataranedotas, 2015-10-17:
According to my data for the Fluence:
Request 0x 22 33 07 to the EVC should give you the heating water temperature +40C

In the Fluence the cooling water for the charger is the same as for the Inverter block, I don't have it confirmed but might be request 0x 22 33 F2 or 0x 22 33 F6

kaimaik, 2015-10-17:
Der Kangoo und der Fluence haben den gleichen Motor.
Das Ladegerät und der Inverter sollten auch gleich sein .
Die Temperaturen  vom Inverter block schwanken je nach fahrweise.Wenn viel Gas gegeben wird gehen die Temperaturen sehr stark nach unten.
Bei einer längeren Fahrt mit wenig Gas verändert sich die Temperatur nur wenig.

Dann müsste 0x 22 33 F6 die Temperatur vom Inverter sein

yoh-there, 2015-11-16:
@avataranedotas Are you interested in building this part in CanZE? ("No" is a valid answer!!) It could be based on the technical charging activity. And I think if build should be put in the (new) technical fragment anyway.

I have closed #87 which in my opinion should be on the same screen due to lack of interest. So far I have not seen much drive to implement any of this either.......

avataranedotas, 2015-11-16:
So basically the idea is to get the temperatures for the various cooling circuits? Is that correct?

yoh-there, 2015-11-16:
I think yes, preferably with the fan status added. Zoe is probably a tad different because it can cool the batteries through a separate evaporator on the CLIMA. On the other hand, you guys have the peltiers.

kaimaik, 2015-11-16:
Ich benötige die Temperaturen dringend.
Für Heizung und das Ladegerät mit dem ganz großen Lüfter

yoh-there, 2015-11-16:
@kaimaik Not to be rude, but I don't think a lot of people understand WHY this is needed . And we really have no resources at all, it is all a labour of love. Why is this useful information?

kaimaik, 2015-11-16:
Tut mir leid wenn das unhöflich rübergekommen ist.
War keine absicht.Sorry
Ich habe bei mir eine Standheizung eingebaut.
Diese Temperauren möchte ich gern überwachen.
Wann die Boardheizung einschaltet und wieviel die Standheizung dazu heizt.

Die Temperaturen vom Ladegerät sind mir auch wichtig.
Ich habe gern ein kühles Ladegerät : -)

yoh-there, 2015-11-16:
OK, makes more sense, Standheizung, interesting. I assume electrical, externally powered then.

 And I was not accusing you of being unhöflich, I was trying to avoid being rude myself!!! :-)

kaimaik, 2015-11-16:
alles ok :-)

kaimaik, 2015-11-16:
Typ2 für den Kangoo oder Fluence
[image]

yoh-there, 2015-11-16:
Sie nützen die normale Type2 auch für Standheizung? Dass müssen sie noch mal am einem tag erklären bitte ;-)

kaimaik, 2015-11-16:
>
> Sie nützen die normale Type2 auch für Standheizung? Dass müssen sie 
> noch mal am einem tag erklären bitte ;-)
Nein.aber möglich !

Typ2 ist zum laden.

Standheizung ist mit Diesel (Webasto)

kaimaik, 2015-11-16:
[image]

fesch, 2015-11-16:
Dans einzig Diesel das ich in meinem Zoé hatte war für meinen Rasenmäher ... und dabei bin ich mir doch echt dämlich vorgekommen als ich mit dem Elektrowagen zur Tanke fuhr um den Kanister auf zu füllen ;-) ;-D

kaimaik, 2015-11-16:
ja das kenne ich :-)

kaimaik, 2015-11-16:
aber besser als frieren

avataranedotas, 2015-11-21:
I now have confirmed data in the Fluence for the following:
-Evaporator temperature
-Water temperature used for heating
-External ambient temperature
-Internal ambient temperature

I will continue to find more confirmed data

avataranedotas, 2015-11-21:
I also have unconfirmed data about the battery cooling fans speed.

yoh-there, I don't have much time for the App, but I can provide you with the addresses of the data above.

yoh-there, 2015-11-21:
Well I bet that covers all. If you added it to the CanZE spreadsheet model or can give it to me spreadsheet-style, in would not be crazy much work anymore I guess. The battery temp I won't take on as the heatmap covers that nicely.

kaimaik, 2015-11-21:
Jetzt fehlt noch die Temperature Ladegerät (Wasserkühler) oder habe ich da was übersehen ?

yoh-there, 2015-11-22:
The experimental temperatures just didn't make it into today's release but it is committed to the master branch. I will close this issue now, but feel free to reopen next week if things are wrong (which I am sure they are as it is totally untested ;-) )

closed 2015-11-22 by yoh-there

yoh-there, 2015-12-01:
Hello @kaimaik . Any feedback on this issue? If it all works, we could promote it from "experimental" to "technical"?

kaimaik, 2015-12-01:
Hallo Alexandre,
bei mir funktioniert grad CanZE nicht.
Bob will einen Kangoo leihen und schaue wo das Problem mit den Daten liegt

yoh-there, 2015-12-01:
Oh d'uh. I knew that. Sorry.

kaimaik, 2015-12-01:
Das wird schon.
Bob ist da ganz zuversichtlich.
Ich auch :-)

kaimaik, 2015-12-07:
Die neue Version liefert wieder Daten :-)
Top !

Bei der neuen Version werden die Temperaturen nich richtig angezeigt.
wie kann ich die ganzen Frames "sehen" ?
Beim Icar Soft 907 werden die Temperaturen angezeigt.
Es muss doch möglich sein diese Parameter auszulesen und in CanZE einzubauen.