fesch/CanZE #108: response from ELM

solmoller, 2015-10-18:
My ELM has been working flawlessly, but in one of the later versions, the check for ELM compatibility is getting a bit too narrow

ELM327.java:
        // only do version control at a full reset
        if (toughness <= 1 && !response.toUpperCase().contains("V1.4") && !response.toUpperCase().contains("V1.5")) {
            MainActivity.toast("ELM is not a version 1.4 or 1.5 [" + response.replace("\r", "<cr>").replace(" ", "<sp>") + "]");
            return false;

My ELM answers 'innocar' so it's not accepted. I've modified the code and it dows work fine when I jump through the above code line.

As we cannot implement all the possible responses from different dongles, I suggest we remove the textual recognition of proper ELM's

Henrik

yoh-there, 2015-10-18:
> response.replace("\r", "").replace(" ", "") + "]");

You are right, I will fix that immediately. Please pull in 15 minutes

yoh-there, 2015-10-18:
BTW, I disagree on removing it altogether, sorry. The scary shit we had
with the dash and cruise control going bezerks.......

fesch, 2015-10-18:
2 other ideas:
- make the verification an "option"
- pull the list of valid strings from the internet (using a dedicated button in the settings)

yoh-there, 2015-10-18:
Done in ELM and ELMexperimental.

I think we should concentrate on a serious testing routine. Even the 1.4 and 1.5 strings are no guarantee whatsoever. We need to think a bit on that, as the current command routines are privately embedded in the ELM classes and the query loop is normally "running". I would like to avoid recoding it all, even as copy/paste, as tweaking will invalidate the testing routines. Can we discuss how to approach that please?

If we still opt for remotely pulled strings, of course that should be integrated with firmware versions pulls (and pushes?)

solmoller, 2015-10-18:
We could alter to a warning: "1.4 reported: should work", "innocar reported: should work" "ELM 2.0 reported: Unknown device, please report if this works"

yoh-there, 2015-10-18:
@solmoller Oh that is a great idea. I will do that. For now, please let me know if your problem is solved

solmoller, 2015-10-18:
ELM327.java : Error:(19, 8) error: ELM327 is not abstract and does not override abstract method join() in Device

yoh-there, 2015-10-18:
I submitted with a SYNTAX ERROR????? Hold on please.

yoh-there, 2015-10-18:
Re-pull and try again please

solmoller, 2015-10-18:
It's not a syntax error - the error is that we should either define the class as abstract or add
    @Override
    public void join() throws InterruptedException {
        
    }

to be added or

yoh-there, 2015-10-18:
Committed. The driver, after all this work, still sometimes gets "out of sync" I noticed.

closed 2015-10-18 by yoh-there