fesch/CanZE #121: Miles

andyfras, 2015-10-21:
Please may we have the option to display miles instead of km?

yoh-there, 2015-10-21:
Really? ;-) I though that was a thing only in use on the other side of the BIG pond? You guys didn't move on????

Seriously, IMHO it is a valid, but at this moment a bit low prio request. But you can help moving this forward a bit. We don't know if for "miles set car" the speed and Odometer is simply redone by the dash and everything on the CANbus is metric (I assume so). If that is true, you should see in CanZE still totally valid ranges, speeds etc, in metric. Can you confirm this is the case?

A note though: The text oriented screens could be relatively easily modified. The widgets probably not.

andyfras, 2015-10-21:
Yes, I can confirm that the current CanZE reading of 27km for range shows as 16 miles on the dashboard.

I appreciate that this is a low priority.

yoh-there, 2015-10-21:
OK, thanks for figuring that. We'll keep this one open.

yoh-there, 2015-10-23:
Implemented in settings and text screens. The widgets, not done, and not sure how easy or hard that is. For planning purposes, this should already be much better for the UK drivers I think?

@fesch Leaving this open as I don't know abut the widgets. Feel free to implement or not (yet). Leaving this open, you can decide when to close.

fesch, 2015-10-24:
@yoh-there: In order to get that done in the widgets too, I needed to add some generic code to the Field class, that does the conversion over there, so the activity will no longer be in charger of the conversion. Probably you have to recheck your code ...

fesch, 2015-10-24:
:-(

closed 2015-10-24 by fesch