fesch/CanZE #122: Hardware Issue - Acer Iconia B1

andyfras, 2015-10-21:
I have tried build 2015-10-13 and 2015-10-18 - in each case it shows ' ELM is not responding'.

Konnwei KV902 Adapter

Acer Iconia Android version 4.1.2

It works OK on my Moto G phone.

yoh-there, 2015-10-21:
If you use more than one device, please realize the KV902 will only remember ONE paired phone. Switching to another device you need to re-pair it all over again. Let us know please if this is the issue. It is the sort of issue that is completely out of our control.

For the time being I am assuming it is exactly this and I am marking this issue as Open (so you can easily respond) and WontFix.

andyfras, 2015-10-21:
Both devices connect to the KV902 without re-pairing, so it seems that my KV902 can remember more than 1 paired device.

fesch, 2015-10-21:
... which I can confirm for the MaxiScan too.

yoh-there, 2015-10-21:
In that case, for the moment I can only advice to install Bluetooth Terminal https://play.google.com/store/apps/details?id=Qwerty.BluetoothTerminal on the Acer, connect to the device (you need to set Append Newline on in the settings screen) and send ATZ Please state the response here.

In the current source (and the build of upcoming Sunday) there will be a screen in the app showing the low level conversation, so as an alternative to Bluetooth Terminal and/or Android Studio, you could opt to wait for that one.

andyfras, 2015-10-21:
Thanks. I'll wait for the next build.

yoh-there, 2015-10-26:
Dear @andyfras . Can you please update us on this issue? Thanks

andyfras, 2015-10-26:
I loaded the latest version (25-10-15), but it still fails with "ELM is not responding" and a "Timeout on [atws]" error.

fesch, 2015-10-26:
Did you try the terminal thing yoh-there asked you for?

yoh-there, 2015-10-26:
What it means is that Bluetooth communication is established, but the ELM it is not talking. I have no idea what to think of this. In any case it is out of the realm of our influence. As said above you can try to use Bluetooth terminal. I am regretfully closing this issue and will include the Acer Iconia B1 as a possibly problematic device on the web page.

closed 2015-10-26 by yoh-there