fesch/CanZE #125: ELM stability

fesch, 2015-10-23:
Today, the ELM run very stable. Once connected, it did not loose the connection.

yoh-there, 2015-10-23:
I totally agree. Things are ROCK stable. Going in and out of focus is NO problem at all, after a bit of hick-ups, everything resets nicely and comes back online. I actually think we can consider lowering the log level. Two remarks though:

1. If the BT connection itself is lost, things do not recover at all. I think that needs to be fixed as it can happen pretty easily. Try switching off the ELM for a few seconds and back on again to test my scenario please.

2. With the fragmenting, closing the app after only having activated the driving screen, gives me a white screen needing another return press. Sometimes, this leads to a black screen and the entire app locks up. To reset I need to press center short, going back to android's main screen, then long, to get to the running applications, then clear that list, killing canze, then start again.

fesch, 2015-10-24:
I have worked this morning on the BT reconnect problem. Waiting for someone to confirm that it has been solved before I close this issue.

yoh-there, 2015-10-24:
Works like a charm! Thank you! Needs a bit of driving around, but Canze recovered from both switching to other applications as well as switching off the ELM. Great work!

closed 2015-10-24 by fesch