fesch/CanZE #135: Export data to CSV on sdcard1

chrismade, 2015-10-24:
> Von:     Henrik Møller Jørgensen <henrik@lillefamilie.dk>
> An:      canze@fisch.lu
> Datum:   24.10.2015 11:02
> Betreff: Data export

 A simple data export option could be working like this:

When data export is enabled, every restart of Canze starts a new csv file on the phone. it is named yyyymmdd_hhmm_odo_Batthealth_outsidetemp.csv
Every 6 seconds a line is added to the file with
Battery%, Battery kWh, ODO*, Bat1temp, speed, GPS coordinates**
 *ODO in file is probably 'since last service', as it appears to have 100 meters resolution

 **GPS coordinates... 

I don't think this would be too hard and it'll open op a numbing amount of vizualization tools.

The file name is a bit long, but it makes it easy to track values that rarely change

There's no need to log more often, and this strategy will keep storage usage in check. I've worked with systems that meter 'intelligently' to conserve data, and IMHO the only acceptable other strategy than fixed time intervals is fixed time intervals with auto-pause :-)
With such a data extract we could get data out of the car and work on vizualization at a much higher speed than compiling and driving for each change.

Further enhancements:

A different file format could be stored when charging the car while the Canze app is stored. A third file could be written to every time the app is started or key is turned, logging ODO, KWh and GPS, effectively forming a drivers log automatically
> >
> > Henrik

As for implementation, I would think the best way to do it would be to add a call to a (static?) logger-class-to-make-method in the queryNextFilter method of each device. As long as some running screen is active this loops through all the queries to make to the car. That call could check for a flag that is being triggered by a timer. Ish ;-)

yoh-there, 2015-10-24:
To avoid going down a closed road: GPS coordinates are not available on the main CANbus, and might be available on the Multimedia CANbus. The latter cannot be read without specific hardware modifications, so I would regard that as a no-go for the average user. Get those from the phone-tablet is my strong advice.

solmoller, 2015-10-24:
Exactly, GPS should be taken from Android

HarmOtten, 2015-10-25:
Do you mean the SDcard of the car with the TOMTOM cards, or from the Android device? If you mean the SDcard in the car, then I would recommend to use the USB-connector for storage, if possible...

chrismade, 2015-10-26:
I mean SDcard in the [our extra] Android device (e.g. Phone, Tablet) - AFAIK TomTom SDcards have their own partition table and copy protection etc - but you brought up a valid point - the built-in Navi/Multimedia display is also an Android device - perfectly suited for our purposes (fixed installed, power supply, connected to CAN-BUS etc) - but unfortunately not yet rooted, no custom rom, not intended to run any user-apps like CanZE  - until we have solved this unsolveable issue - we need to bring an _extra_ Android device, and when we talk SDcard or GPS we mean those of our extra device

yoh-there, 2015-10-26:
Modifying R-Link is something that has come up in past a lot. It's not reasonably doable. I will create (and immediately close) an issue about it right now so we can refer to that when (not if) the discussion comes up again.

yoh-there, 2015-11-05:
I admit I have not been following progress on logging closely, so I do not know "where we are", other than the implementation of the time based application queue, which is great. So please take the following remark with that lack of actual knowledge.

I think we would have to compile a list of "reasonably loggable parameters" (a subset of "Fields" initialisation, plus maybe GPS), normalize their naming (thank god their ID is already normalized) and make them selectable for logging. As the obvious format is CSV, I would strongy advise to always write a first line with the normalized names (or ID's) of the now-being-logged fields.

As for the "reasonably loggable parameters", I am probably speaking for @HarmOtten when I say it should include SOC, mean battery compartment temperature (I think I need to add that in Fields) and Max Charging/Regen.

And while we're at it (ducking), it would be a really nice option to upload logs OTA :-o

fesch, 2015-11-16:
I have modified the already present class to build up a DebugLogger which is now in the actual release and working. Log files are saved to the /CanZE/ folder.
Datalogging itself could be hooked up to the CanzeDataSource, feeding the DataLogger, which will be all for the functional part.
Still nee some settings to let the user chose on what to log...

fesch, 2015-12-20:
This is being implemented. So it is no longer an issue --> closing

closed 2015-12-20 by fesch