fesch/CanZE #139: read (write) data stream in raw format instead of bluetooth

chrismade, 2015-10-24:
using the android simulator for dev I found that bluetooth is not supported - and I haven't found any other way to simulate the data stream from the car. Maybe there is a low-effort smart way to read data from file instead of bluetooth (which has been previously recorded in the real car) - at least for dev purposes (simulator), maybe there are also other use-cases

yoh-there, 2015-10-24:
Canze depends on two types of CANbus frames: free frames (which we need to trick the ELM in giving those to us), and diagnostics, which are on demand.

Thing is, the ELM is very slow and can not eve remotely record the car's data stream, so you're loosing serious testing there too, meaning you'd need at least a Due to record the data. Not a huge deal, some of us have that hardware and we could share a few real life files.

It would be not VERY hard to replay raw frame data though it would need an entirely new, rather weird driver (store frames per ID while they are played back and fetch current value when queried by the app). And we'd still loose the diagnostics messages. And we would also loose testing the entire BT and device handling.

A more interesting approach would be (though not solving the android emulator issue!) to build a car-plus-ELM/Due-Bluetooth emulator. It could run on a Due with a BT module. And it would be quite the challenge to build! I think I have spend about what that would cost on the soles of my shoes, going back and forth to the car ;-)

Let me be blunt (Dutch): I am not going to do either. It seems not worth the effort, though I would support anyone taking up building either (and I can make a few CRTD formatted log files using the Due, GVRET and Savvycan)

fesch, 2015-10-25:
I think this is too much overhead and impossible in some other ways too. For me it is not an issue to move to the car. I know that for the ELM this is absolutely needed, but hey, we got it done until now too :-D

closed 2015-10-25 by fesch