fesch/CanZE #140: Formal and legal disclaimer first time App is open

avataranedotas, 2015-10-25:
I would suggest that on the first time the app is open (after installation) a disclaimer be presented to the user that would have to click on checkbox and accept button before the app starts connecting to the car.

fesch, 2015-10-25:
I will do that ... but please, could anyone provide me the text to display? I must admit that this kind of texting is none of things I'm really good in ...

yoh-there, 2015-10-25:
I'd simply copy the text from the Download page of the website :-)

fesch, 2015-10-25:
It's in the code now. To test it:
- Open CanZE
- Goto settings
- Clear the cached data
- Quite CanZE while in the Settings-Screen
- Restart CanZE 
- Voilé!

closed 2015-10-25 by fesch

yoh-there, 2015-10-25:
Erm no? What do you mean with quitting CanZE while in the settings screen? I even tried clicking "center" button to get to Android, then clear last active apps, nothing gives me the disclaimer.

fesch, 2015-10-25:
Quit CanZE using the "show apps" button (on mine it is a simple white square), then slide CanZE away to close it.

yoh-there, 2015-10-25:
OK, yes, did that, a "pkill -9 CanZE" equivalent ;-) But....

I do get "You first need to adjust....", but no disclaimer.

fesch, 2015-10-25:
I will try on some other device later ...
[image]

yoh-there, 2015-10-25:
I like the "No" button ;-) But no, doesn't work here, sorry

yoh-there, 2015-10-25:
Owwww, when I normally close an "unset" CanZE now, it crashes (as in, seems to close normally, then a "CanZE stopped" popup. So it has to do wit the failing disclaimer as everything is OK after settings have been saved.

fesch, 2015-10-25:
Very strange ... any exception in the LogCat? Does the warning pop-up if you unset the "Safe Driving Mode"?

yoh-there, 2015-10-25:
Works. Half-compiled crap on my end. Apologies :-/

fesch, 2015-10-25:
Ufff ...

yoh-there, 2015-10-25:
Well, maybe. Could be your last "code move" commit did the trick. Works after that one, thanks!

avataranedotas, 2015-10-25:
I think the part "By using this software you are interfering with your car and doing that with hardware and software beyond your control, created by a loose team of interested amateurs in this field. Any car is a possibly lethal piece of machinery and you might hurt or kill yourself or others using it, or even paying attention to the displays instead of watching the road." should also be in, with bigger text than the formal part.

fesch, 2015-10-26:
[image]

avataranedotas, 2015-10-26:
Perfect!