fesch/CanZE #144: CanZE for R-Link

yoh-there, 2015-10-26:
it is often suggested CanZE should be implemented on R-Link. Here is an early (and if I my say rather blunt) example from the German forum:

> In the light of the recent "stagefright" vulnerability of Android (and also R-Link), wouldn't it
> be easier to hack the OS itself, so one can directly use the CAN processor on the mainboard
> and the internal display? Up to now, I haven't found any sign of hacking attempts of the
> R-Link system itself. There must be lots of other open doors, given that it is based on the
> ancient Android 2.3.

On the face of it, the argument is not unreasonable:
- The display and processing power is there;
- It is at least connected to the MM CANbus (and maybe the main CANbus, we don't know);
- No additional devices needed, no mounting brackets for an additional tablet or phone.

The, there are also strong arguments and uncertainties against:
- R-Link is, as said, based on a totally outdated Android version (2.3);
- We would probably be forced to use R-Link UI design elements and rules (not entirely illogical BTW), requiring new learning and cutting flexibility options;
- We don't even know if R-Link is connected to the main CANbus;
- There is no publicly available tool chain to develop apps fro R-Link;
- If there is a closed tool chain and if we could get our hands on it, it is probably a step back in productivity (Eclipse instead of IntelliJ based)
- We would most probably have to deal with the whims of both Renault and TomTom;
- We would have to be much more careful not to break the host platform;
- Testing would most probably totally car-bound.

I can't see this happening really, unless there is explicit demand from Renault. I doubt they are interested and given the effort needed we'd have to even talk monetary compensation. I dare to state that I speak for most of the team-members / developers when I say that they are not willing to either risk breaking R-Link or buying / learning / maintaining a specific tool chain and emulator / hardware for this project.

closed 2015-10-26 by yoh-there

fesch, 2015-10-28:
Just some added information:
R-Link uses different parts licensed under the GPL license and thus it's source should be available in request. The last page of the "Copyright Screen" on the R-Link indicates this too. 
So, who ever wants it, can grad it here, it was easy to find:
http://www.tomtom.com/en_gb/gpl/renault_r_link/?WT.Click_Link=home_quick_link

yoh-there, 2015-10-29:
Interesting stuff! I had a quick peek but it seems to me there is nothing there that could help us any further. It seemed mostly the Android core system. It is a huge leap to get from there to a development chain. I stand by my statement above.

yoh-there, 2015-11-01:
I had a discussion with Renault about R-Link development. Won't go in more details but the essence of it is that it not doable and totally depending on Renault France. Furthermore, R-Link does not have  direct link to the main CANbus.

egonalter, 2016-01-06:
Ok, if it's not connected to the main bus, it is probably really moot. I also found by disassable that the new firmware release 2094591 fixes at least some of the stagefright bugs.