fesch/CanZE #148: Use "releases" feature

fesch, 2015-10-27:
Introduced by mail:

And generally just FYI for GitHub. One thing which is often forgotten is
the release tagging. E.g. [here](https://github.com/fesch/CanZE/releases) you can tag a release
and IMHO you should do that, because this way it's easy to find a
release and you can publish the compiled versions there too.

fesch, 2015-10-27:
Tested with the latest build. Will do that next time too ...

closed 2015-10-27 by fesch

yoh-there, 2015-10-27:
You know what? I think that is amazingly cool, and I think you should ONLY do that, linkt from the download page to github release. Hard? I don't think so as we're already in the realm of installing APK's.

Speaking of......

I think we should move to the the step, publishing on the Play Store. Auto updating for the crowd, yay.

fesch, 2015-10-27:
OK, the link is up and running ...