fesch/CanZE #151: PollerThread does not if switching to fast ...

fesch, 2015-10-28:
If switching pausing and resuming the activity is done too fast, the pollerThread will not yet be assigned and therefore it is not being stopped.

1) open CanZE
2) open Consumption activity
3) pause CanZE (BT connection will drop as expected!)
- now be fast please!
4) resume CanZE 
5) pause CanZE (BT connection will not yet have been established and thus the thread is only assigned after the last pause, so it won't be stopped as the onPause event has already finished ...)

fesch, 2015-10-28:
Can't reproduce on Due/Teensy, so it must be some king of ELM issue I think. Will re-check in the car later on ...

but anyhow, the looping code could be extracted from classes and put into the abstract class as it is the same for all device anyway ...

yoh-there, 2015-10-28:
I agree, see current elm code. The loop is very neat and compact now with
the two request functions now separated out.

fesch, 2015-10-28:
I've done some modifications to the different device classes, but I will not commit until tested.

It would thus be nice if no one could touch any file inside the "devices" package for now ...

fesch, 2015-10-28:
Due/Teens is tested (test environment).
--> Will need to go to the car for a ELM test ...

yoh-there, 2015-10-28:
No problem.
Would it not be an idea to implement a small queue mechanism in the MainActivity to handle switching the poller? Or maybe even better if the poller simply quits based on the fields-to-poll queue? It seems to get a bit complicated..

fesch, 2015-10-28:
AFAIK with the recent changes I made, the queue will never be empty, so we could not use that as an indicator. When maximizing a widget, the queue as well as the registered fields should not be touched, so stopping the poller on each pausing activity is no option too.
In fact, it should be stopped only when the entire application is being paused or destroyed, but as Android is not telling me this, I have to figure it out by myself. The actual API does not give any information about this.
http://stackoverflow.com/questions/6224693/application-pause-resume-state

yoh-there, 2015-10-28:
OK, I understand that. Still, there are activities that like the poller stopped, i.e. ELM test. DTC's we MIGHT get away with manipulating the queue for displaying them, but there we need to be able to give "one off", not to be repeated (IE clear DTC) commands too. Please don't forget that with implementing these changes :-).

For the former, from an interface point of view, the best would be to just call a method an onCreate and then forget about it, to avoid issues when closing such a special activity and forgetting to "restore".

For the latter, I think we would need some sort of "addOneShotListener"? What do you think?

fesch, 2015-10-28:
"addOneShotListener" for what? To query a field you mean?
If so, I would like to prefer a "numberOfRequestsToSend" (or something similar). -1 would be "inifinit", any positive number will be decreased until it reaches 0 after which the field will be kicked from the queue.
If we do so, the activities could set the number of requests by their own. For application requested field, this setting will be ignored ...

yoh-there, 2015-10-28:
That would be fine, though I do not understand what you mean with "application requested fields" (as opposed to activities) Maybe the latter is i.e. your speed-safe check? Which btw could be removed if Safe mode is off.

The NumberOfRequest should not be in the fields definition I think, but a separate parameter.

While we're thinking about this, maybe keep in mind the "logging" request. That seems to be something similar, but time bases, i.e. a (set of) queries every x seconds (if the device can keep up). ~Shudder~

fesch, 2015-10-28:
Yes, I have defined two queues:
- on which is activity specific
- the other one is application specific. And you are right, actually the speed is on that list as it will always be monitored (yes, it could be removed if the mode is off .. that is right, I'll do that).

The logging facility was the second reason why I added the application related queue ;-)

yoh-there, 2015-10-28:
I rest my case Sir! Please be patient with me ;-)

And please be kind to i.e. ElmTest (free up the device entirely and keep a handle on the BT serial link)

fesch, 2015-10-28:
The dual queue thing is totally transparent to the activities.
I've modified it now, so that the speed is no longer monitored it safe driving mode is off.
No change to the behavior nor to the method names (Device had an abstract method, which is no longer abstract now), so neither the ElmTest nor the DtcActivity should be concerned about these changes.

fesch, 2015-10-28:
https://www.youtube.com/watch?v=1wdJBCRNpQs

... has to be re-integrated, because the actual version is *not* stable!

yoh-there, 2015-10-28:
Interesting video!

yoh-there, 2015-10-28:
Sorry, that video comment was tongue-in-cheek. What is it supposed to mean???

fesch, 2015-10-28:
Dammit, I've pasted the wrong link ;-)

https://www.youtube.com/watch?v=KOqU9_KW4aI

Just listen ...

yoh-there, 2015-10-28:
Ah I see........

https://www.youtube.com/watch?v=FmFA7jDnk_w

fesch, 2015-10-29:
All modifications are done & tested. No interruption during my drive today!

yoh-there, 2015-10-29:
Sorry to poor water on a drowning man.......

When switching to another App and coming back, things do not restore properly. My impression was this was really solved around the weekend, but that part is very unstable again.

fesch, 2015-10-29:
Will try ... 
The code is now basically the same as before the weekend, but there too, it was not stable for me anyway! Even on a 16km drive, I sometimes lost sync or when switching to another app, things became bad on restoring CanZE. I have to dig somewhat more ... or maybe recode it entirely ...

fesch, 2015-10-29:
I have reviewed the BT code: Correct me if I'm wrong, but as we are no longer pushing any data from the connected OBD device (Due/Teensy) to the Android app, but only working on a "per request" basis, the "ConnectedBluetoothThread" is of no use at all, isn't it?
It just created one more thread and thus increases the complexity of the entire code. We would better trash it and talk immediately to the BT-Manager class, which I have designed as singleton ...

yoh-there, 2015-10-29:
I think we have implicitly decided NOT to send everything we see on the
canbus to Android. It could be done, but it would heavily and unnecessarily
overload the BT connection and the device. For ELM's it would not work at
all (or only under very specific circumstances). So no, we keep the
initiative. Even if we start using GVRET, we would still implement a new
command to stop blurping all the data but just store the last frame per ID
and get them to CanZE on CanZe's request.

Whether inquiring the device is done in a device driver loop, a device
driver queue or through direct control of the activity (testing, DTC's)
needs to controllable, but is indeed irrelevant to the principle.

If saying that means ConnectedBluetoothThread can be tossed, then indeed,
things can be tossed. IMHO.

fesch, 2015-10-29:
Staring, stopping, pausing CanZE: the BT connection closes and open's as expected. Even rotating the device like a helicopter's propeller is no problem. Also smashing it on the ground make the BT connection close (shit! shit! shit! ... wasn't this protective case not supposed to PROTECT the pad???)

I will do some driving tests tomorrow. Can't leave the house right now ...

closed 2015-10-30 by fesch