fesch/CanZE #157: CanZE crashes on Battery section on HTC One Mini

tho-wa, 2015-10-31:
CanZE crashes when opening Battery section on HTC One Mini.
Other Sections are working.

fesch, 2015-10-31:
Ok ... I would need a copy of what Logcat spites out. tho-wa, do you think you could install Android Studio, connect your device, start the Battery activity and the cyop & paste the output from the logcat over here?

Looking at it only from what I read in the title and description, I can't find any grabble entry point  to  start searching for a problem ...

fesch, 2015-11-02:
The report in the Play Store only gives this, which is not much :-(

java.lang.NullPointerException
	at lu.fisch.canze.activities.d.run(Unknown Source)
	at java.lang.Thread.run(Thread.java:864)

yoh-there, 2015-11-02:
Totally useless! I have no idea? SHould it be compiled with other options
to at least ehm....... tell us something? How do even now it was battery?

fesch, 2015-11-03:
I were not able to reproduce with the latest build, so it must be a problem specific to the Sunday build. I have published a new version on the Play Store. Please be patient ... it needs some time to get ready ...

Please re-open the issue if the problems persists ....

closed 2015-11-03 by fesch