fesch/CanZE #158: Research project ELM

yoh-there, 2015-10-31:
We MIGHT be able to tweak the ELM filters that the ATMA command will pass on several (but just the ones we want) free frame types. Suppose for a moment this works and we have a screen using only free frames (braking is one of them). Suppose also a worst case: 4 frames, 10ms intervals.

400 fps, each frame 25 characters, each character 10 bits = 100.000 bps. This is way above the default internal UART of the ELM, but the documentation says it can easily handle that when the internal baud rate is increased.

We would have a super-responsive (Due comparable) system.

Apart from diagnostics (like cell voltages), real time data is next to always available as free frames. 

Thoughts?

fesch, 2015-10-31:
Are you talking about hacking the ELM and putting a new firmware on it?

yoh-there, 2015-10-31:
No, programming the filters smarter, advanced AT stuff

yoh-there, 2015-11-01:
I re-read the documentation. Somehow the idea had settled in my brain that it was possible to set multiple pattern filters, but this is not the case. Without that, there is not a lot we we can optimize further.

I have contemplated on a few alternatives, and I think we should stick to the current implementation, though I do believe we might be able to improve a bit on the queues. But that is for another issue.

Closing this idea.

closed 2015-11-01 by yoh-there