fesch/CanZE #161: Drawable/progressbar_canze_green missing

solmoller, 2015-11-01:
Finally found out how to add the upstream fork - thanks for the instructions!

Unfortunately there is a module that is nos GIT'ed

Error:(54, 40) No resource found that matches the given name (at 'progressDrawable' with value '@drawable/progressbar_canze_green').

yoh-there, 2015-11-01:
- will look in green now. May I forgot to add it to git
- please indicate where there are omissions in the Wiki with respect to forking / problems, https://github.com/fesch/CanZE/wiki/3.-Collaborate-%28one-time-setup%29 last paragraph and https://github.com/fesch/CanZE/wiki/5.-Issues-while-developing first item

yoh-there, 2015-11-01:
Green bar indeed forgotten to add to git. Done, committed, pushed and merged.

closed 2015-11-01 by yoh-there