fesch/CanZE #166: Smarter memory usage for huge tables

yoh-there, 2015-11-02:
I was wondering if we could "smarten up" the big lookup tables. Now, the fields code is basically in memory twice: once as a huge string in program memory, once in RAM once initialized. This is a waste of resources. Can't we keep the data in static program memory? If so, what would be the best way of doing that?

One idea is to change the initialisation code to use static, hardcoded (read: generated from Gcalc) data, as this not only gets rid of the double storage, but also automatically uses string interning, saving a ton of memory. https://blog.codecentric.de/en/2012/03/save-memory-by-using-string-intern-in-java/

Side note: I believe we should simplify the Field and Fields classes substantially anyway. I don't think anyone uses the format representations or any of the strings for that matter, with exception of the units. 

I am asking as I am working on the implementation of the DTC readout, which again requires a substantial and growing number of in itself completely static final strings that requires only a lookup through a 3 byte code and probably an ECU code too.

Reading in from an external source (file) is only superficially solving the issue. The data would still be on the device twice.

This is a low priority issue in my opinion.

fesch, 2015-11-02:
Wait ... I think this is only partially true!  In fact, the "big string" is bound to a method, not to the class. 

This means, that when method "fillStatic()" is being called, the string is being put on the stack. As soon as the method terminated, the stack is being cleaned up again. So what is left, is what is contained in the ArrayList ...

yoh-there, 2015-11-02:
Is it put on the stack? You know more about this than me, but I thought as it was being an Object static final, the bigstring would simply sit in "program memory", as in, what is being loaded at invocation, and had it not been static final, it would be implicitly copied to the heap at initialization (as explicitly the ArrayList is, isn't it?).

If you're saying all this stuff is being cleared out when the method is terminated (program and it's static final data) I rest my case, well, a bit ;-)

fesch, 2015-11-02:
The big string that comes from the spreadsheet is not static *and* located inside the method, so when the class is being loaded and even when the first object is being created, it will not yet be in memory. 

Only when the method "fillStatic()" is being called, the string will be loaded in memory, the field array will be filled and when the method terminated, the big string will be kicked of from the stack again. I'm pretty sure about this ;-)

Now for the "intern()" method: Thanks, I didn't know it exists. Apparently, as the "fillStatic()" method is only being called once, it does not slow down the app.

closed 2015-11-02 by fesch

yoh-there, 2015-11-02:
Learning new things every day :-) But you're closing too fast :D I thought that the code part of classes were loaded as a unit, not method by method. But I believe you of course when you say that is not being the case with it's associated data.

How would you then advice to implement the DTC table? While a Hashmap with an integer key would be the natural candidate, I am still not sure if this is smart. It feels like complete overkill to fill the entire hashmap from a bigString, only to lookup a couple of descriptions. The code is called only when asking for DTC's (so it should not be initialized application wide at all), and basically it is a lookup from a 3.5 byte int to a string for descriptional purposes only.

fesch, 2015-11-02:
OK, I now see what you mean by "the code is loaded as a unit". I must admit, that I'm not sure about that part. As it is interpreted code, there are different mechanisms. Anyway, the Java Memory Management is not that easy to understand. As far as I have read, *for non-static methods, all the parameters and local variables go onto the stack - and whenever there is a concrete invocation of that method, we get a new stack-frame associated with it.* That would mean, that the big string is pushed on the stack only when the method is being invoked. I would need to make some tests to be sure about this. 

Interesting thing ... ;-)

For the DTC table, first of all I did not yet capture how this works, so giving an advice is somewhat difficult, but if I got it right, another option would be to use a flat file and to parse this for a certain value. This is the classical I/O versus memory problem. Either you are fast and waste some memory, or you are slow because of readings from the disk (OK, here we have flash drives ... wouldn't that be OK?)

yoh-there, 2015-11-02:
Man I am glad I an teach something back!! And please come back to me after
your tests, because like you, I am pretty sure on this, but........ you
never know!!! Here goes:

An object's data is created on the heap, not the stack. On the stack is
indeed your local variable, which is only a pointer to the data of the
object. And then there is the "balloon" model. Every object (the real data,
the balloon) is referenced (tied down) with one or more ropes (pointers).
Once all pointers are destroyed (all ropes are cut), the balloon can float
away, which is what the garbage collector does.

fesch, 2015-11-02:
Just did some very quick tests with CanZE (more tomorrow ...) and monitored the memory usage in three different situations:

fillStatic called (normal start) = 7.73 MB
no fillStatic called = 6.04 MB
big string commented out = 6.41 MB

That's not the result I expected ... at least not for the last part :-/

yoh-there, 2015-11-02:
Eh??????