fesch/CanZE #167: License of the main launcher icon ...

fesch, 2015-11-02:
This image is all around the Internet. I can't remember where exactly where it has been taken from, but I think it has come from some official Renault site, but I can't find it right now. For the same reason, I cannot indicate the license of this image right now, but as it can be  found on many sites, I suppose it may be at least usable without modifications for non-commercial  use. I would be grateful for any more precise information ...

yoh-there, 2015-11-02:
I think indeed it is a Renault image, i.e. go here.....

http://www.renault.nl/showroom/personenautos/twingo/twingo/#veries-en-uitrust

and see "other cars" at the far bottom.

There are three roads we can take here
1. Ignore the issue and hope for the best. It would be violating the GPL no doubt and it would vialote Renault's (or the photographer's) copyright for sure. Chances of getting "caught" would probably be 0.0001%

2. Address the issue. The only option here would be to ask Renault. My contacts are useless here and to be totally honest, I am not going to ask them in this "premature" phase of inside contacts. They would need to climb higher in the corporate (legal) tree for us and it is something that I simply cannot expect them to do

3. Shrug it off and make a new icon.

Here's my proposal, based on the fact that should Renault object, they still need to give us a fair chance to correct it: do nothing (option 1), in the mean time, in the "ask for help", and maybe on a few fora, ask for a volonteer to make us someting (option 3).

solmoller, 2015-11-02:
Option 3! There's gotta be  someone within reach

fesch, 2015-11-04:
While waiting for someone being more competent in the graphical domain, I've painted a new icon myself based on the previous one. No, it's not a copy, I started from a blank sheet of A4 paper (yes, this thin white thing) ... scanned and photoshopped a bit, but that's mainly everything ...
[image]

fesch, 2015-11-04:
The initial issue is solved, so I close it, but we keep on looking for some new icon I think ... maybe without a Zoé ...

closed 2015-11-04 by fesch