fesch/CanZE #170: CanZe stürzt ab

kaimaik, 2015-11-02:
Hallo,
bei Battery und Consumption stürzt das Programm ab.
mit dem Tablet Samsung Tab A mit LTE
und  dem Handy Samsung Duos 

Build:2015.11.01 @18:02

Die anderen Funktionen laufen soweit

fesch, 2015-11-02:
Ohne weitere Angaben wie z.B. einer Debugger Ausgabe, kann ich da leider nicht viel sage.

Eventuell CanZE via Google Play Store installieren und den Crash erneut provozieren. Im besten Fall wird dann ein Report an Google gesendet den ich sehen und in welchem dann hoffentlich eine nützliche Fehlermeldung steht.

yoh-there, 2015-11-02:
Bob, please realize kaimaik drives a Kangoo Z.E.. Could it have to do with
i.e. asking compartment temperature sensors beyond #4, which is the number
of sensors in the Kangoo & Fluence? Maybe ending up beyond the string
length?

fesch, 2015-11-02:
Ahhh I didn't know that!

In that case we must look for a NullPointerException. Mostly when a field is being lookuped for a given car, but when it is not found, null will be returned. So we must make always some null-test before continuing ...

yoh-there, 2015-11-02:
Thinking on how you meant that. In my screens I always check in addListener
if the field exists and "Toast" if not. Saved me probably 1000 crashed ;-)

When running, I don't think a null can come back to onFieldUpdateEvent could
it? It's paramter is ALWAYS a valid instance of FIeld I hope?

kaimaik, 2015-11-03:
Nun da gibt es keine Debugger Ausgabe.

Sowie ich den Button gedrückt habe,dauert es so 1-2 Sekunden dann 
verabschiedet sich CanZE

jragg, 2015-11-03:
Confirmed at Build 2015.11.01 @ 18:02:
immediately crashes at:
Battery, Consumption and
Experimental:
TEMP(BOB)
BattTempAkt(BOB)
Leafspy(BOB)
Tachoactivity(BOB)

Most of them are working in previous versions

XIAOMI RedmiNote 4G, Android MIUI 5.8.20/decuro , Android 4.4.4

Thanks and keep up ur good work !!!

J.R.

HarmOtten, 2015-11-03:
Same here with same build as jragg mentioned. I installed it via the google playstore.
I have a "astar" tablet.
It still worked on the 2015.10.25 apk version.

fesch, 2015-11-03:
I see this in the developer console:

    java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String lu.fisch.canze.widgets.b.c()' on a null object reference
    	at lu.fisch.canze.activities.d.run(Unknown Source)
    	at java.lang.Thread.run(Thread.java:818)

fesch, 2015-11-03:
OK, that must be somehow related to the widgets ... I'check it ...

fesch, 2015-11-03:
I were able to reproduce on some other device ...

fesch, 2015-11-03:
I were not able to reproduce with the latest build, so it must be a problem specific to the Sunday build. I have published a new version on the Play Store. Please be patient ... it needs some time to get ready ...

closed 2015-11-03 by fesch

jragg, 2015-11-03:
now seems all o.k.!
Thanks!

PR #178 (2015-11-03): Small changes in Driving, Charging (closed #170) and dtcs
Small changes in Driving, Charging (closed #170) and dtcs
commit 56b1dbdc (2015-11-03): Merge pull request #178 from yoh-there/master

Small changes in Driving, Charging (closed #170) and dtcs
commit bcf0d1a1 (2015-11-03): Small changes in Driving, Charging (closed #170) and dtcs