fesch/CanZE #175: TPMS coding/initializing

iamlien, 2015-11-03:
Zoe does only have space in memory for 4 TPMS sensors. When living in a country where winter tyres are a necessity, this means two visits to the shop a year for recoding the TPMS sensors, alternatively driving around without working TPMS and flashing warnings in the dash.

Being able to initialize the TPMS with Canze would be a awesome. Possible?

Cheers!

yoh-there, 2015-11-03:
We would need to have a general idea on how that is done. Does it require fiddling with a CLIP tool (Renault's diagnostic suite) to do that?

iamlien, 2015-11-04:
I really have no idea, but I will ask the mechanic when I go in next week.

yoh-there, 2015-11-04:
I have looked a bit into it and it seems TPMS is really again something invented anew from the entire network stack up :-( :-( 433 Mz transmission, the same technology that is used for remote light switches in your house, and LF reception (only used to force a transmission). And yes, usually hand held garage tools to link it all up. If the latter is true, we cannot do anything in CanZE to implement. If it is done through the CANBUS *) , we need to have a "sniffer session" while a mechanic does that, which leads me to..........

There is nothing I can check on it: TPMS is mandatory in Europe starting mid 2014, and my car is not equipped, build September 2013. I think I am one of three team members who build sniffer hardware and all of us own pre-TPMS cars.

Yes, if you could please ask as many questions as you can next week, that might be helpful. Somehow I am slightly pessimistic.

*) If I were a gambling man I'd say there is a new hardware version of the BCM computer and implemented there.

yoh-there, 2015-11-16:
I am closing this issue. We have zero information on how this should be done.

bjaolsen, 2018-11-09:
This is possible to do using ddt4all and ELM327, it does however require you to have the ids for both sets of tires. The ids can also be read out from the unit using the same software, so once you have had the workshop do this once you can basically do it yourself afterwards.

In ddt4all the TPMS information is found in the UBIC/BFR section.
I used the screen TPMS_learning_ID_Pressure which allowed me to read and write all 4 ids at once

yoh-there, 2018-11-10:
Thank you for that information. It's a bit unfortunate that both @fesch and myself don't have TPMS equiped cars as to replay such sessions. People are basically dying to have this done without dealer interference.

bjaolsen, 2018-11-10:
I'll try to do it again and capture the CAN traffic if that may help

yoh-there, 2018-11-11:
That would be cool indeed and maybe we can put that in the experimental section first. It would require some UI tweaking though, something like "save current TMPS set as...".

yoh-there, 2019-01-13:
State of affairs. To do this requires at least three technical challenges:
- we need to ensure we can do multi-frame ISOTP message sending. This probably works for the generic dongles, but is untested code in the CanSee dongle that has our full focus at the moment;
- I need to get DDT4All up and running on my PC (dongle connectivity issues);
- I need to have a car with TPMS available and experiment with doing this in DDT4All while checking the traffic with Savvycan. Quite the contraption to set up and haul to a volunteer. Especially in winter ;-)

yoh-there, 2019-01-14:
ISOTP sending long frames for CanSee dongle: done but untested
DDT4All installed and up and running: check! (Activated Android Auto with it)
Car with TPMS: volunteer close to me found.
Stay tuned

bjaolsen, 2019-01-14:
<CAN BaudRate="500000" Multipoint="1">
      <SendId>
        <CANId Value="1861" />
      </SendId>
      <ReceiveId>
        <CANId Value="1893" />
      </ReceiveId>
    </CAN>

<Request Name="Write_TPMS_Configuration_4">
        <DossierMaintenabilite />
        <ManuelSend />
        <Sent>
          <SentBytes>3B5D000000000000000000000000</SentBytes>
          <DataItem Name="CodeIdentite(1)" FirstByte="3" />
          <DataItem Name="CodeIdentite(2)" FirstByte="6" />
          <DataItem Name="CodeIdentite(3)" FirstByte="9" />
          <DataItem Name="CodeIdentite(4)" FirstByte="12" />
        </Sent>
        <Received MinBytes="2">
          <ReplyBytes>7B5D</ReplyBytes>
        </Received>
      </Request>

      <Request Name="Read_TPMS_configuration">
        <Sent>
          <SentBytes>2171</SentBytes>
        </Sent>
        <Received MinBytes="42">

<ReplyBytes>617100000000000000000000000000000000000000000000000000000000000000000000000000000000</ReplyBytes>
          <DataItem Name="PRESENCE_SSPP" Ref="1" FirstByte="3" />
          <DataItem Name="TramesDebug" Ref="1" FirstByte="3" BitOffset="1"
/>
          <DataItem Name="CodeIdentite(1)_(0)" Ref="1" FirstByte="4" />
          <DataItem Name="CodeIdentite(2)_(0)" Ref="1" FirstByte="7" />
          <DataItem Name="CodeIdentite(3)_(0)" Ref="1" FirstByte="10" />
          <DataItem Name="CodeIdentite(4)_(0)" Ref="1" FirstByte="13" />
          <DataItem Name="Para_Defaut_Total" Ref="1" FirstByte="28" />
          <DataItem Name="Para_Tempo_Emission_Roulage" Ref="1"
FirstByte="29" />
          <DataItem Name="Para_Seuil_Vitesse_Basse" Ref="1" FirstByte="31"
/>
          <DataItem Name="Para_Defaut_Batterie" Ref="1" FirstByte="33" />
          <DataItem Name="Seuil_Temperature_Gestion_Pile_Faible" Ref="1"
FirstByte="34" />
          <DataItem
Name="Pression_gonflage_preconisee_basse_vitesse_a_l_avant" Ref="1"
FirstByte="35" />
          <DataItem
Name="Pression_gonflage_preconisee_basse_vitesse_a_l_arriere" Ref="1"
FirstByte="36" />
          <DataItem
Name="Pression_gonflage_preconisee_haute_vitesse_a_l_avant" Ref="1"
FirstByte="37" />
          <DataItem
Name="Pression_gonflage_preconisee_haute_vitesse_a_l_arriere" Ref="1"
FirstByte="38" />
          <DataItem Name="PARA_TEMPO_VITESSE_MANQUANTE" Ref="1"
FirstByte="39" />
          <DataItem Name="PARA_LOC_LEARNING_DEFAULT" Ref="1" FirstByte="40"
/>
          <DataItem Name="PARA_LOC_LEARNING_CONVERGENCE_TIMEOUT" Ref="1"
FirstByte="41" />
        </Received>
      </Request>

yoh-there, 2019-01-14:
Thank you sir :-)