fesch/CanZE #188: Use Ecus and Frames in Fields and device implementations

yoh-there, 2015-11-06:
Initialize Ecus and Frames when Fields is initialized, link frame object to each field instead of timing.

Replace isIsoTp with ID>0x700 (this is safe and according to the OBD2 standard)

Replace custom code to find ECU id's to the methods of the Ecu(s) class.

Note: this issue is a placeholder. Current optimizations should be run for at least a week before implementing this.

yoh-there, 2015-11-07:
Ecu(s) implemented in ELM and DTC. If I read the code correctly, there is no needs (or place) to implement this in BobDue. I assume the from2to table is handled in the Due code for ISO-TP frames

fesch, 2015-11-07:
Correct for the actual version of my firmware ...

closed 2015-11-08 by yoh-there

fesch, 2015-11-10:
Saw this message today: frame does not exist:0x722

0x722 is LINSCH I think ...

yoh-there, 2015-11-10:
Oops. 722 is indeed linsch (which does not exist). Removed it from Fields (and the spreadsheet). Thanks!