fesch/CanZE #197: ELM testing is done

yoh-there, 2015-11-08:
ELM testing (now still hidden under Playground (Jeroen) is done.

You can give it a spin. It should be performed in a "started" car! It performs

- device initialisation. If any command fails, it states what command and stops further processing. This should catch 99% if the incompatible devices;

- It does a software version request to the instrument panel. The outcome is not important, if there IS an output, that should be 99% OK

- It does a free frame request to the parking brake. Again, the outcome is not important, if there IS an output, that should be 99% OK

If a few of you could give it a spin, I can "promote" it to the main screen (or somewhere in the settings).

fesch, 2015-11-08:
Mine looks like this:

[image]

Is that a good or a bad sign?

yoh-there, 2015-11-08:
It is a good sign! Maybe I should add a length calculation saying: OK! And
a final conclusion

yoh-there, 2015-11-08:
Thanks for the feedback, I improved what is reported a lot, including a final conclusion if things look good.

closed 2015-11-08 by yoh-there